Livox-SDK/Livox-SDK #4: livox mid-40 failed to start sampling

CaptD, 2019-02-01:
Hi, I just received the livox mid-40 lidar. I followed the instructions to connect the lidar and my laptop to the same router via ethernet. The lidar is powered by a power supply and the sync wire is not plugged in. However, there is no pointcloud data. 

I tried both linux 16.04 and windows livox viewer, and they both failed. Screenshots are attached. How can I diagnose this issue?

[image]
[image]

Thanks.

Livox-SDK, 2019-02-02:
NOTE: Please replace the broadcast code lists in the main.c for both LiDAR sample and Hub sample with the broadcast code of your devices before running. You can find the following code section in main.c:

#define BROADCAST_CODE_LIST_SIZE 3
char *broadcast_code_list[BROADCAST_CODE_LIST_SIZE] = {
    "00000000000002",
    "00000000000003",
    "00000000000004"
};

Livox-SDK, 2019-02-02:
Power supply may be insufficient，it‘s recommended to use a more powerful power supply！
Thanks！

CaptD, 2019-02-02:
Thanks for quick response. Will try both later and come back to follow up.

Livox-SDK, 2019-02-02:
CaptD，the voltage is between 12v and 14v！

CaptD, 2019-02-02:
Got it. I used a power supply, and tried 12 and 16v. Still no help..

CaptD, 2019-02-10:
[image]
Hi. I followed your instructions, and there are points now. However, the points detected have a weird pattern, shown in the picture above. And on left, the status is warning 0x40000040....

Livox-SDK, 2019-02-11:
[image]
Hi,CaptD, please remove the protective film on the optical windows，this may be the cause of warning！

Livox-SDK, 2019-02-11:
About the pointcloud pattern，you can referrence this document:
https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/Point%20cloud%20characteristics.pdf

CaptD, 2019-02-11:
Thanks a lot. The warning is gone after I removed the protective screen. As for the point cloud pattern, I carefully read the doc you shared. My understanding is that this is an expected phenomenon, and we need to filter out the pattern points during post process by filtering out near distance points. Is my understanding correct?

CaptD, 2019-02-11:
Sorry my previous understanding is wrong. Now I get it. And I got it working in Ubuntu 16.04 after updating broadcast ID. Thanks for your help~

closed 2019-02-11 by CaptD