Livox-SDK/Livox-SDK #8: Pi Builds failing

upworkEdisonBing, 2019-05-14:
Compiling when outside the 64bit shell yields errors relating to apr1.h and off_64_t mentioned in the first post

Is it possible to compile 32bit version of the livox-sdk? Reason being that Pi3 camera is inaccessible in the ds64 shell 

follow the guide for installation on ubuntu on my raspberry pi, but running the make command gives med the following error about unknown type name 'off64_t'

pi@raspberrypi:~/livox/Livox-SDK/build $ make
[ 84%] Built target livox_sdk_static
[ 88%] Building C object sample/hub/CMakeFiles/hub_sample.dir/main.c.o
In file included from /usr/include/apr-1.0/apr_general.h:28,
                 from /home/pi/livox/Livox-SDK/sample/hub/main.c:33:
/usr/include/apr-1.0/apr.h:358:10: error: unknown type name ‘off64_t’
 typedef  off64_t           apr_off_t;
          ^~~~~~~
sample/hub/CMakeFiles/hub_sample.dir/build.make:62: recipe for target 'sample/hub/CMakeFiles/hub_sample.dir/main.c.o' failed
make[2]: *** [sample/hub/CMakeFiles/hub_sample.dir/main.c.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'sample/hub/CMakeFiles/hub_sample.dir/all' failed
make[1]: *** [sample/hub/CMakeFiles/hub_sample.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Is there a specific way for installation on a raspberry pi or what to do?

Livox-SDK, 2019-05-16:
Hi,
  upworkEdisonBing, we will be releasing a new version at this weekend that will resolve this "off64_t" error!

fjserrano, 2019-05-20:
Hi, while their release the new version I have a workaround. On the file main.c on the "Livox-SDK/sample/lidar/main.c" for compiling examples:

  I put before the line "#include <apr_general.h>" the next line "typedef long long  off64_t;"

  The header of file is like this:

`#include <stdio.h>
#include <stdlib.h>
#ifdef WIN32
#include <windows.h>
#else
#include <unistd.h>
#endif
#include <string.h>

#include <inttypes.h>

typedef long long  off64_t;
#include <apr_general.h>
#include <apr_getopt.h>
#include "livox_sdk.h"`

  With this I can compile example and running the software example.

Best Regards.

Livox-SDK, 2019-05-22:
Dear user:
We have updated the wiki page about how to install Livox-SDK in Linux 32 bit system. This can work in RPi 3B+.  Please uninstall your libapr first before installing apr from source which is mentioned in [wiki page](https://github.com/Livox-SDK/Livox-SDK/wiki/The-installation-of-Livox-SDK-on-Linux-32-bit-system-and-Mac).

closed 2019-07-29 by Livox-SDK