Livox-SDK/Livox-SDK #11: Get information and Data from SDK with no static callbacks C functions

fjserrano, 2019-05-28:
Hi.

  I am developing a system with your Lidars. I have my applications in C++ and all code have Objects with their respective Classes.

  I want a class for controlling the Lidar, but the SDK all the information has the Callbacks for Get information and data of Lidar and on C++ in a Class with not static functions I can't configure Callbacks.

  I don't see any methods on SDK to connect lidar, get information or get data without Callbacks, It is correct??. 

  **Do you have any plan for have a C++ Class of methods or methods to get information directly from a function returning the structures requested from a specific Lidar??**

  My system have one defined Lidar for device and have the specfied broadcast_code for connect and Static IP, and I want to connect directly without broadcasting and waiting packets for connect.

  In a first approach my class looks like this and fails compiling on "SetDeviceStateUpdateCallback":

Lidar.h:

```

#include "livox_sdk.h"

typedef enum {
	kDeviceStateDisconnect = 0,
	kDeviceStateConnect = 1,
	kDeviceStateSampling = 2,
} DeviceState;

class Lidar
{
private: //Properties
	string sTAG = "Lidar";

	string sCode = "XXXXXX7006003XX";

	uint8_t uiHandle = 0;
	uint8_t uiDeviceState = kDeviceStateDisconnect;
	DeviceInfo xDeviceInfo;
public: //Properties

private: //Method
	void vOnDeviceChange(const DeviceInfo *info, DeviceEvent type);

public: //Method
	Lidar();
	~Lidar();
};
```

Lidar.cpp:
```
#include "Lidar.h"

void Lidar::vOnDeviceChange(const DeviceInfo * xDeviceInfo, DeviceEvent xDeviceEventType)
{
	try {
		if (xDeviceInfo != NULL) {
			if (uiHandle == xDeviceInfo->handle) {
				if (xDeviceEventType == kEventConnect) {
					//QueryDeviceInformation(this->uiHandle, Lidar::vOnDeviceInformation, NULL);
					this->uiDeviceState = kDeviceStateConnect;
					this->xDeviceInfo = *xDeviceInfo;
				}
				else if (xDeviceEventType == kEventDisconnect) {
					this->uiDeviceState = kDeviceStateDisconnect;
				}
				else if (xDeviceEventType == kEventStateChange) {
					this->xDeviceInfo = *xDeviceInfo;
				}

				if (this->uiDeviceState == kDeviceStateConnect) {
					if (this->xDeviceInfo.state == kLidarStateNormal) {
						this->uiDeviceState = kDeviceStateSampling;
					}
				}
			}
			else {
				
			}
		}else{
			
		}
	}
	catch (std::exception& e) {
		std::cout << e.what() << '\n' << std::endl;
	}
}

Lidar::Lidar()
{
	try {
		/** Initialize Livox-SDK. */
		if (Init()) {
			Constants::vShowLog(this->sTAG, "Livox SDK has been initialized.");
			LivoxSdkVersion xSDKVersion;
			GetLivoxSdkVersion(&xSDKVersion);

			//Constants::vShowLog(this->sTAG, "Livox SDK Version" + std::to_string(xSDKVersion.major) + "." + std::to_string(xSDKVersion.minor) + "." + std::to_string(xSDKVersion.patch));

			/** Set the callback function called when device state change,
			* which means connection/disconnection and changing of LiDAR state.
			*/
			//SetDeviceStateUpdateCallback(&Lidar::vOnDeviceChange);

			//SetDeviceStateUpdateCallback([this](const DeviceInfo * xDeviceInfo, DeviceEvent xDeviceEventType) {vOnDeviceChange(xDeviceInfo, xDeviceEventType); });

			auto vCallbackOnDeviceChange = std::bind(&Lidar::vOnDeviceChange, this);
			SetDeviceStateUpdateCallback(vCallbackOnDeviceChange);

			/** Start the device discovering routine. */
			if (!Start()) {
				Uninit();
			}
			else {
				usleep(1000000);

				bool bRes = false;
				bRes = AddLidarToConnect(sCode.c_str(), &uiHandle);
				if (bRes == kStatusSuccess) {
					/** Set the point cloud data for a specific Livox LiDAR. */
					//SetDataCallback(uiHandle, Lidar::vGetLidarData);
					this->uiHandle = uiHandle;
					this->uiDeviceState = kDeviceStateDisconnect;
				}
			}
		}
		else {

		}
	}
	catch (std::exception& e) {
		std::cout << e.what() << '\n' << std::endl;
	}
}

Lidar::~Lidar()
{
}
```

Best Regards.

Livox-SDK, 2019-06-04:
Hi fjserrano.
We are planning to add C++ interface based on existed C interface or C++ core.
Maybe it will cause an error when using the lambda expression based on the C interface.

fjserrano, 2019-07-15:
Hi. 

  Do you have road map for C++ interface??.

Best Regards.

Livox-SDK, 2019-08-02:
Hi, C++ samples are available now.
Best Regards.

closed 2019-08-21 by Livox-SDK