Livox-SDK/Livox-SDK #12: PTP Sync

shomin, 2019-06-18:
I can't get the Mid-40 to achieve a PTP sync.  Is there any way to introspect what could be the problem?  The device is on the same subnet as a PTP master.  A pc on the same subnet is able to PTP sync using ptp4l (linuxptp) successfully.  Do I need to do anything else to trigger the sync?  All I ever get is no-sync timestamps (ns since power-on).

Livox-SDK, 2019-06-20:
We have tested ptpd project as master. You can refer to this: [ptpd](https://github.com/Livox-SDK/ptpd)

shomin, 2019-06-20:
Ok, I will try ptpd specifically.  Do I need to issue any API commands to trigger a sync?  Is there any feedback on PTP status?  All I see in the docs is that PTP is highest priority.  If a PTP master exists on the local network, should I expect the Mid-40 to automatically sync and start broadcast timestamps appropriately?

Livox-SDK, 2019-06-28:
You don't have to trigger the IEEE1588 sync. There is a byte in data package to indicate the type of synchronization.

shomin, 2019-07-03:
Is the device expecting the PTP master to be at a particular IP address?  I am running the ptpd that you suggested.  It is able to serve as the master for another computer running ptp.  I can see the request and the sync from the other computer, but I see nothing coming from the Mid-40.

Livox-SDK, 2019-07-04:
Mid-40 can only ptp sync after the handshake connection, you can use sdk or livox viewer to connect lidar and try again.

closed 2019-10-15 by Livox-SDK

calligaris, 2022-12-14:
Hi, in the message above, "handshake" refers to the PTP handshake or the Livox protocol handshake?