Livox-SDK/Livox-SDK #19: Livox SDK on Win10

mi4869, 2019-11-25:
Hi,

I have followed instructions to compile the Livox SDK on Win10. After connecting my Mid-40 successfully, I run lidar_sample.exe following Step 4.2.2, and it shows the following figures. Is it normal? I dont know why the programme automatically stops.
 
[image]

ososinski, 2019-11-28:
The count seems to increase "0" corresponds to the lidar handle and "2670000" - "2900000" is the packet count, suggesting that the lidar works fine

The reason it stops is that the sample is designed to work for a number of seconds and stop.

If you go through the source code you will find the exact number.

mi4869, 2019-11-28:
> The count seems to increase "0" corresponds to the lidar handle and "2670000" - "2900000" is the packet count, suggesting that the lidar works fine
> 
> The reason it stops is that the sample is designed to work for a number of seconds and stop.
> 
> If you go through the source code you will find the exact number.
> 
> If you want to visually confirm that the lidar works, use [Livox Viewer.exe ](https://www.livoxtech.com/mid-40-and-mid-100/downloads)

Thanks ososinski, it works using Livox Viewer. If I want to run [Livox ROS Driver](https://github.com/Livox-SDK/livox_ros_driver), do I need to run any files under Livox-SDK?

ososinski, 2019-11-28:
Just follow the instructions for the ros-sdk installer, then

to run the visualisation:
`roslaunch livox_ros_driver livox_lidar_rviz.launch`

to run as standalone ros package ie for loam-livox
`roslaunch livox_ros_driver livox_lidar.launch`

Hope your project will be up and running soon ^_^

mi4869, 2019-11-29:
I think the instruction is for Linux, but I am using Win10 as I need to use some softwares applicable for Win10 instead of Linux. It seems that I cannot build Livox ROS Driver correctly by following the instruction.

ososinski, 2019-11-29:
Sorry I do not have any experience with ROS on windows, but the ROS wiki should have some hints on how to build packages

Livox-SDK, 2019-12-16:
Livox ROS driver don't support windows

closed 2020-09-07 by Livox-SDK