Livox-SDK/Livox-SDK #21: 如何获取正确的时间戳？

wdweidao, 2019-12-04:
我用的是mid-40,我自己写了个脚本，抓取雷达数据，但是抓到的时间戳转换为日期格式是1970年的时间，请问如何获取正确的时间？

wdweidao, 2019-12-04:
能否在发送采样命令时，通过cmd命令将正确的时间戳写入雷达

Livox-SDK, 2019-12-16:
在没有外部同步时钟的情况下，mid40的时间戳是从0开始增加的绝对时间，单位是ns

qjw2bqn, 2020-01-15:
hello，i export the .csv data from livox viewer.  Some column which field name is Timestamp  value is 855488071920，then i translate to utc format, the value is 1997/2/6 16:9:23.How to set the correct time on horizon.Other words,how to '同步时钟'

Grandit, 2020-08-05:
> hello，i export the .csv data from livox viewer. Some column which field name is Timestamp value is 855488071920，then i translate to utc format, the value is 1997/2/6 16:9:23.How to set the correct time on horizon.Other words,how to '同步时钟'

I have the same problem too， and any commet has about it, @Livox-SDK ?

closed 2020-09-07 by Livox-SDK