Livox-SDK/Livox-SDK #22: Interferences between multiple sensors

marcel1991, 2019-12-26:
I use 2 livox mid-40 sensors that scan at the same time. When I get the scanned output I see there are a lot of wrong measurements, e.g. very far values etc. When the sensors scan the environment one after another and not at the same time everything works fine. So I assume some reflections from one sensor go into into the other sensor leading to wrong measurements.

Am I doing something wrong? Is there a way to filter out these wrong measurement? 

This is really a big issue. The sensors are supposed to work together...

Livox-SDK, 2019-12-30:
If you set different handle to the same DataCallback cb when use SetDataCallback interface:
  `void SetDataCallback(uint8_t handle, DataCallback cb, void *client_data);` 
eg:
```
void GetLidarData(uint8_t handle, LivoxEthPacket *data, uint32_t data_num, void *client_data) { 
 //.....
}
SetDataCallback(handle1, GetLidarData, NULL);
SetDataCallback(handle2, GetLidarData, NULL);`
```
 
In this case, DataCallback cb will be called in different lidar thread. if you didn't lock `point_cloud_buffer`, multiple sensors' output may mix together, eg:
```
void GetLidarData(uint8_t handle, LivoxEthPacket *data, uint32_t data_num, void *client_data) {
  if (data) {
    std::unique_lock<std::mutex> lock(mtx);
    // ...
    // point_cloud_buffer = process(data);
    // ...
  }
}
```

Livox-SDK, 2020-01-02:
if it doesn't work, could you please describe usage scenario and operating steps in detail and send an e-mail to us dev@livoxtech.com, thanks!

closed 2020-09-07 by Livox-SDK