Livox-SDK/Livox-SDK #29: Setting Static IP

dleliuhin, 2020-02-07:
Hi Livox team,
I'm one of Chief specialists in JSC NIIAS. There was a need to change the default IP address for the  Mid-40 and Horizon. 
Relying on _4.1 General CMD Set 0x08 Configure Static/Dynamic IP_  can change only 4th block of IP such as: 192.168.150.***. But we need to set up 192.168.***.***. How can I change it? Had tried to change in Livox Viewer but there is no such option. Sending the command package did not bring the desired results. 
Ask for help in resolving this issue.

Livox-SDK, 2020-02-10:
Currently, Mid-40 is not supported to set IP to `192.168.***.***`   
But Horizon can change default IP to `192.168.***.***` through Livox Viewer or Sending the command package.

dleliuhin, 2020-02-10:
Thank you!

Can you answer on my questions: https://groups.google.com/forum/#!topic/livox-lidars/Lthh71wLTcM ?
or there:
Used the firmware from https://github.com/Livox-SDK/Special-Firmwares-for-Livox-LiDARs  and a couple of questions arose:
1) Is it possible to install 3 firmware on a lidar at once or do I need to choose only one?
2) As far as I understand, the firmware is valid only for the Mid-40/100 models. Will the same be released for Horizon and Tele-15? Especially for Horizon, as it cuts off points up to 1.5 meters.
3) Is it possible to flash lidar using Livox-SDK? If not, will there be such an opportunity in the future?
4) Will there be firmware that will cut off the ground points? If not, what algorithms would you recommend?

Livox-SDK, 2020-02-11:
1. choose only one
2. Yes, at present, Special-Firmwares is only available for Mid40 / 100 modules.
Horizon / Tele-15 already supports multiple return and tag filtering features in the official firmware.
3. Horizon and Mid40 / 100 currently have no plans to make Livox-SDK support flash LiDAR. For subsequent products, we have plans to make Livox-SDK support flash LiDAR.
4. no, the firmware will not support this function. This function can be processed by some higher-level algorithms. such as:
You can use [RANSAC PLANE MODEL](http://www.pointclouds.org/documentation/tutorials/planar_segmentation.php) for plane fitting, and then filter based on plane area + vector direction.

closed 2020-09-07 by Livox-SDK