Livox-SDK/Livox-SDK #32: Livox lidar ghost points in point cloud

nithi89, 2020-03-06:
Hi I am trying to use the Livox mid-100 series lidar. The point cloud has ghost points as you can see in the point cloud attached.

pointcloud image: https://www.dropbox.com/s/4d9galew50uf3bd/35_ptcld_ghostpts.png?dl=0

corresponding camera image: https://www.dropbox.com/s/4d9galew50uf3bd/35_ptcld_ghostpts.png?dl=0

All objects in the scene are stationary (including cars). Can you please explain this behavior and also how to fix it?

Thank you.

dleliuhin, 2020-03-06:
May be there are noise points due to high humidity. You can apply some noise removal filters for 3D point cloud data. See there: 
https://github.com/search?q=lidar+noise

Livox-SDK, 2020-03-17:
Hi @nithi89. To analyze this behavior，can you provide us with pointcloud's lvx file (use Livox Viewer to record within 20s) and corresponding camera image.

nithi89, 2020-03-17:
Hi, I realized the problem was a software error and not with the lidar, it
is fixed now. Thank you.

closed 2020-04-14 by Livox-SDK

meng-zha, 2021-07-28:
@nithi89 Hi, I met the similar problem with mid-100 lidar that there are noisy points along the border of obstacle. Do you remember what the software error you met was at that time?