Livox-SDK/Livox-SDK #33: sample/Lidar的例子中，雷达被遮挡的时候会调用OnLidarErrorStatusCallback么

qulee, 2020-03-11:
printf("dirty_warn : %u\n", message->lidar_error_code.dirty_warn);

Livox-SDK, 2020-03-16:
您好，雷达被遮挡的时候OnLidarErrorStatusCallback不会被调用。
在点云采样过程中，雷达被遮挡后有，SDK有两种方式会通知用户
（1）参照：[SDK通信协议](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol) 3.1 Package Format ，点云的数据包格式 status_code相应的字段dirty_warn 会被置位为1. 
（2）雷达被遮挡后，SDK会触发SetDeviceStateUpdateCallback()设置的回调函数。
例如sample/lidar的例子中，会触发void OnDeviceInfoChange(const DeviceInfo *info, DeviceEvent type)，其中 info->status 中也会记录相关的脏污预警信息。

qulee, 2020-03-16:
Thanks 

 

 例如sample/lidar的例子中，会触发void OnDeviceInfoChange(const DeviceInfo *info, DeviceEvent type)，其中 info-&gt;status 中也会记录相关的脏污预警信息。

closed 2020-04-14 by Livox-SDK