Livox-SDK/Livox-SDK #35: Warning when operating Livox Horizon

xmba15, 2020-03-24:
Hi. I got the following warning when I connected Livox Horizon to see the point clouds on the Viewer:
0x40000040, which shows Dirty (as can be seen in the attached images). I already removed the protective film from the lidar.
[image]
[image]

Livox-SDK, 2020-03-25:
Hi, @xmba15.
Could you confirm that the distance between the object and the Livox Horizon is less than 1m ? If so, Horizon may trigger dirty warning.
If you have questions about the quality of the lidar, It's recommended to issue at [ livox lidars google group](https://groups.google.com/forum/#!forum/livox-lidars)

xmba15, 2020-03-25:
Hi, thank you for your reply.
The warning actually disappears as I point the lidar away from the nearby (less than 1m) obstacles.
I will post questions (if any) on google group next time.
Thanks for your support.

closed 2020-03-25 by xmba15