Livox-SDK/Livox-SDK #37: Question: how to integrate the Livox-SDK into a OOP structure?.

lasdasdas, 2020-04-02:
Hi. I am trying to integrate the drivers into my own software and I am having some problems.

The use of *global* objects for the callbacks (i.e. ```SetDeviceStateUpdateCallback``` ```SetBroadcastCallback``` ```LidarStartSampling``` ...) is not viable in my case. I have to put all the global objects as non static members of a class. I have seen that some of the callbacks use a void pointer to pass *generic* data structures to the callbacks, however some of the others don't. I have tried std::bind (which would be perfect for this case) .
For instance:
```c++
void OnDeviceInfoChangeCustom(MyCustomClass* a, const DeviceInfo *info, DeviceEvent type) {
//...
}
//And then

SetDeviceStateUpdateCallback(std::function<...> (std::bind(OnDeviceInfoChangeCustom, this, _1, _2)
```

However this is not working either. Partially because SetDeviceStateUpdateCallback expects the type DeviceStateUpdateCallback only to cast it later into a more flexible std::function (just since today)/boost::function).

Do you have any recommendation for this situation?. 
Thank you very much.

Livox-SDK, 2020-04-10:
Hi @lasdasdas ，
Thank you for your proposal ! 
We will plan to provide C++ interface which make it easier to intergrate into OOP structure.

fjserrano, 2020-04-13:
Hi @Livox-SDK.

  I can suggest apart of make a std::function and std::bind for a Interface OOP (I thinks is the best) other method to implement in C++. You can make a Interface by polling data where we need to request the new data from SDK for any handle, only is a idea.

Best Regards.

Livox-SDK, 2020-04-16:
Hi @fjserrano ,

Thank you for your suggestion ! we will take it into consideration carefully.

lasdasdas, 2020-04-16:
It is a good idea. Also, some of the functions have a free void pointer that can be used to pass anything you need (DeviceInformationCallback, SetDataCallback) however others do not ( DeviceStateUpdateCallback...) Maybe @Livox-SDK can explain the reasons for this interface decision. If all callbacks had a void pointer, using classes would be trivial as the class object itself could be passed as ```this```

closed 2020-09-07 by Livox-SDK