Livox-SDK/Livox-SDK #44: setdatacallback无法触发回调

zhangchaoltt, 2020-04-16:
按照sample例子，setdatacallback无法触发回调，操作系统Ubuntu18.04
[image]

qulee, 2020-04-16:
是不是没建立连接   试试Lidar_lvx_sample的例子 看看能不能读到点云数据

zhangchaoltt, 2020-04-16:
Lidar_lvx_sample例子执行完创建了一个.lvx文件，但是用livox_viewer工具加载这个文件报错： Reading lvx file failed! This file may have been corrupted

zhangchaoltt, 2020-04-16:
难道livox_viewer不支持ubuntu18.04？

qulee, 2020-04-16:
有没有文件执行的截图啊 感觉你说的情况应该是没有建立socket连接 可能是收到设备广播后卡在建立连接那了  是不是网络的问题啊 IP设置之类的

zhangchaoltt, 2020-04-17:
@qulee LivoxPoint 没有时间戳属性吗？那如何做运动补偿呢

qulee, 2020-04-17:
时间戳在 LivoxEthPacket 里面吧

zhangchaoltt, 2020-04-17:
一个LivoxEthPacket 数据包含多个 LivoxPoint吗，LivoxPoint是点云中的一个点吗? 那LivoxEthPacket有时间戳，那下面的LivoxPoint时间戳咋计算

Livox-SDK, 2020-04-17:
您好， @zhangchaoltt 
1. 如果您是用 SDK V2.0.0 及以上版本运行， livox_viewer 需要 0.7.0 及以上版本才能正确解析 lvx文件格式
2. LivoxEthPacket 数据包含多个 LivoxPoint，每个LivoxPoint的时间间隔均匀分布，以Mid40为例，采样率为100 kHz, 则每个点云的时间间隔为 10 us,  具体内容可以参照[Livox SDK 通信协议](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol) 3.2章节

zhangchaoltt, 2020-04-17:
@Livox-SDK 你好，LivoxEthPacket数据包含多个LivoxPoint，也就是一个packet下面的所有LivoxPoint都用同一个时间戳吧？ 另外一般多少个LivoxEthPacket包凑成一张点云啊？

zhangchaoltt, 2020-04-17:
我们现在使用场景是1秒输出10张点云数据到channel上

closed 2020-09-07 by Livox-SDK