Livox-SDK/Livox-SDK #46: raspberry

Jweiiii, 2020-04-27:
您好，在树莓派下cmake，有错误，请问需要改哪里
[image]

liang-tang, 2020-04-28:
没安装cmake吗?

Jweiiii, 2020-04-28:
安装了
pi@raspberrypi:~ $ cmake -version
cmake version 3.13.4
CMake suite maintained and supported by Kitware (kitware.com/cmake).

liang-tang, 2020-04-28:
哪个版本的树莓派？

Jweiiii, 2020-04-28:
Raspberry Pi 2 Model B Rev 1.1

liang-tang, 2020-04-28:
https://github.com/liang-tang/Livox-SDK/commit/522aab4db249b9b12418687758a06f4ae2957c6d

liang-tang, 2020-04-28:
建议换成树莓派3b或者4b，安装64位操作系统，执行下面的脚本安装环境
https://github.com/liang-tang/Livox-SDK/blob/master/script/install_pi.sh

liang-tang, 2020-04-28:
https://docs.google.com/document/d/14nue8maXYg1HSuOTEA1Ccmx5Bd9_OfFGLZPchpHolGg/edit?usp=sharing

Jweiiii, 2020-04-28:
好的，谢谢啦

closed 2020-09-07 by Livox-SDK