Livox-SDK/Livox-SDK #47: mac环境下，执行make命令的时候发生以下错误

jimiechen, 2020-04-30:
mac环境下，执行make命令的时候发生以下错误，只能ubuntu环境吗，不支持mac编译吗

[ 72%] Linking CXX executable lidar_lvx_sample
[ 72%] Built target lidar_lvx_sample
Scanning dependencies of target hub_sample_cc
[ 74%] Building CXX object sample_cc/hub/CMakeFiles/hub_sample_cc.dir/lds_hub.cpp.o
[ 76%] Building CXX object sample_cc/hub/CMakeFiles/hub_sample_cc.dir/main.cpp.o
[ 79%] Linking CXX executable hub_sample_cc
[ 79%] Built target hub_sample_cc
Scanning dependencies of target lidar_sample_cc
[ 81%] Building CXX object sample_cc/lidar/CMakeFiles/lidar_sample_cc.dir/main.cpp.o
[ 83%] Building CXX object sample_cc/lidar/CMakeFiles/lidar_sample_cc.dir/lds_lidar.cpp.o
[ 86%] Linking CXX executable lidar_sample_cc
[ 86%] Built target lidar_sample_cc
Scanning dependencies of target trouble_shooting
[ 88%] Building CXX object sample_cc/trouble_shooting/CMakeFiles/trouble_shooting.dir/main.cpp.o
[ 90%] Linking CXX executable trouble_shooting
[ 90%] Built target trouble_shooting
Scanning dependencies of target lidar_utc_sync
[ 93%] Building CXX object sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/main.cpp.o
[ 95%] Building CXX object sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/lds_lidar.cpp.o
[ 97%] Building CXX object sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/synchro.cpp.o
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:36: error: use of undeclared identifier 'B460800'; did you mean 'BR460800'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                   ^~~~~~~
                                   BR460800
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:59:3: note: 'BR460800' declared here
  BR460800,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:45: error: use of undeclared identifier 'B500000'; did you mean 'BR500000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                            ^~~~~~~
                                            BR500000
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:60:3: note: 'BR500000' declared here
  BR500000,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:54: error: use of undeclared identifier 'B576000'; did you mean 'BR576000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                     ^~~~~~~
                                                     BR576000
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:61:3: note: 'BR576000' declared here
  BR576000,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:62: error: use of undeclared identifier 'B921600'; did you mean 'BR921600'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                             ^~~~~~~
                                                             BR921600
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:62:3: note: 'BR921600' declared here
  BR921600,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:70: error: use of undeclared identifier 'B1152000'; did you mean 'BR1152000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                                     ^~~~~~~~
                                                                     BR1152000
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:63:3: note: 'BR1152000' declared here
  BR1152000,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:80: error: use of undeclared identifier 'B1500000'
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                                               ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:90: error: use of undeclared identifier 'B2000000'; did you mean 'BR2000000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                                                         ^~~~~~~~
                                                                                         BR2000000
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:65:3: note: 'BR2000000' declared here
  BR2000000,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:36: error: use of undeclared identifier 'B2500000'
                                   B2500000, B3000000, B3500000, B4000000\
                                   ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:46: error: use of undeclared identifier 'B3000000'; did you mean 'BR3000000'?
                                   B2500000, B3000000, B3500000, B4000000\
                                             ^~~~~~~~
                                             BR3000000
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:67:3: note: 'BR3000000' declared here
  BR3000000,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:56: error: use of undeclared identifier 'B3500000'
                                   B2500000, B3000000, B3500000, B4000000\
                                                       ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:66: error: use of undeclared identifier 'B4000000'; did you mean 'BR4000000'?
                                   B2500000, B3000000, B3500000, B4000000\
                                                                 ^~~~~~~~
                                                                 BR4000000
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:69:3: note: 'BR4000000' declared here
  BR4000000,
  ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:296:23: error: use of undeclared identifier 'CBAUD'
  options.c_cflag &= ~CBAUD;
                      ^
/Users/nyis/jemoo/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:299:41: warning: format specifies type 'int' but the argument has type 'tcflag_t' (aka 'unsigned long') [-Wformat]
  printf("[Baudrate]: %d %d\r\n", baud, baudrate);
                         ~~             ^~~~~~~~
                         %lu
1 warning and 12 errors generated.
make[2]: *** [sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/synchro.cpp.o] Error 1
make[1]: *** [sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/all] Error 2
make: *** [all] Error 2

jimiechen, 2020-04-30:
Compile Livox SDK
In the Livox SDK directory, run the following commands to compile the project:

git clone https://github.com/Livox-SDK/Livox-SDK.git
cd Livox-SDK/build
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/usr/local/Cellar/apr/1.X.X/libexec/lib/pkgconfig
cmake ..
make
sudo make install

ZionGao, 2020-05-03:
同款BUG
[ 81%] Building CXX object sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/synchro.cpp.o
[ 93%] Built target hub_sample
[ 93%] Built target hub_lvx_sample
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:36: error: use of undeclared identifier 'B460800'; did you mean 'BR460800'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                   ^~~~~~~
                                   BR460800
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:59:3: note: 'BR460800' declared here
  BR460800,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:45: error: use of undeclared identifier 'B500000'; did you mean 'BR500000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                            ^~~~~~~
                                            BR500000
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:60:3: note: 'BR500000' declared here
  BR500000,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:54: error: use of undeclared identifier 'B576000'; did you mean 'BR576000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                     ^~~~~~~
                                                     BR576000
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:61:3: note: 'BR576000' declared here
  BR576000,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:62: error: use of undeclared identifier 'B921600'; did you mean 'BR921600'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                             ^~~~~~~
                                                             BR921600
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:62:3: note: 'BR921600' declared here
  BR921600,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:70: error: use of undeclared identifier 'B1152000'; did you mean 'BR1152000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                                     ^~~~~~~~
                                                                     BR1152000
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:63:3: note: 'BR1152000' declared here
  BR1152000,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:80: error: use of undeclared identifier 'B1500000'
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                                               ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:273:90: error: use of undeclared identifier 'B2000000'; did you mean 'BR2000000'?
                                   B460800, B500000, B576000,B921600,B1152000, B1500000, B2000000,\
                                                                                         ^~~~~~~~
                                                                                         BR2000000
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:65:3: note: 'BR2000000' declared here
  BR2000000,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:36: error: use of undeclared identifier 'B2500000'
                                   B2500000, B3000000, B3500000, B4000000\
                                   ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:46: error: use of undeclared identifier 'B3000000'; did you mean 'BR3000000'?
                                   B2500000, B3000000, B3500000, B4000000\
                                             ^~~~~~~~
                                             BR3000000
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:67:3: note: 'BR3000000' declared here
  BR3000000,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:56: error: use of undeclared identifier 'B3500000'
                                   B2500000, B3000000, B3500000, B4000000\
                                                       ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:274:66: error: use of undeclared identifier 'B4000000'; did you mean 'BR4000000'?
                                   B2500000, B3000000, B3500000, B4000000\
                                                                 ^~~~~~~~
                                                                 BR4000000
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.h:69:3: note: 'BR4000000' declared here
  BR4000000,
  ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:296:23: error: use of undeclared identifier 'CBAUD'
  options.c_cflag &= ~CBAUD;
                      ^
/Users/gao/CLionProjects/Livox-SDK/sample_cc/lidar_utc_sync/synchro.cpp:299:41: warning: format specifies type 'int' but the argument has type 'tcflag_t' (aka 'unsigned long') [-Wformat]
  printf("[Baudrate]: %d %d\r\n", baud, baudrate);
                         ~~             ^~~~~~~~
                         %lu
1 warning and 12 errors generated.
make[2]: *** [sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/synchro.cpp.o] Error 1
make[1]: *** [sample_cc/lidar_utc_sync/CMakeFiles/lidar_utc_sync.dir/all] Error 2
make: *** [all] Error 2

Livox-SDK, 2020-05-21:
您好，
     目前 mac 环境还不支持 lidar_utc_sync 的例程，可以先将 Livox-SDK/CMakeLists.txt 第24 行注释：
`#add_subdirectory(sample_cc/lidar_utc_sync)`
我们将在后续版本中支持 mac 版本的 lidar_utc_sync

mar2194, 2020-07-16:
I fixed the first issue by changing lines 269-275 in synchro.cpp to:
`/* Sets up the port parameters */`
`int Synchro::Setup(enum BaudRate baud, enum Parity parity) {`
`  static uint32_t baud_map[19] = {\`
                                   `BR2400, BR4800, BR9600, BR19200, BR38400, BR57600,BR115200, BR230400,\`
                                   `BR460800, BR500000, BR576000,BR921600,BR1152000, BR1500000, BR2000000,\`
                                   `BR2500000, BR3000000, BR3500000, BR4000000\`
                                 `};`

The second issue happens when compiling on a Mac because CBAUD is a Linux extension to the POSIX Terminal I/O definitions. The fix is to use the standard POSIX Terminal I/O functions to set the serial input and output BAUD rates (i.e., if CBAUD is undefined). Starting at line 295 in synchro.cpp:
` /* Set boadrate */`
`  #ifdef CBAUD`
  	`options.c_cflag &= ~CBAUD;`
  	`baudrate = baud_map[baud];`
  	`options.c_cflag |= baudrate;`
`  #else`
	`baudrate = baud_map[baud];`
	`cfsetispeed( &options, baudrate );`
	`cfsetospeed( &options, baudrate );`
`  #endif`

closed 2020-09-07 by Livox-SDK