Livox-SDK/Livox-SDK #49: Horizon的点云数据格式与Mid系统不一样？？

shenshen927, 2020-05-09:
Horizon与Mid系列的点云数据格式不一样么？我采用SDK中lidar_lvx_sample这个例子，将接受到的点云数据（x，y，z，r）保存到文本文件中（代码如下），采用其他工具打开该文本文件，可以收集并展示出更多更密集的点云视图。但是用相同的方法接受并保存Horizon的数据就不行，得到的图是个立方体（到处都是反射点，感觉乱掉了）如下图所示。请问，Horizon的点云数据格式是不是与Mid不同，该如何保存x、y、z、r 四个值呢？谢谢。
void LvxFileHandle::SaveTxtFile(std::list<LvxBasePackDetail> &point_list)
{
    std::unique_ptr<char[]> buffer(new char[WRITE_BUFFER_LEN]);
    uint64_t curpos = 0;
    for (std::list<LvxBasePackDetail>::iterator iter = point_list.begin(); iter != point_list.end(); iter++)
    {
          for (size_t i = 0; i < 100; i++)
          {
         	int x = ((LivoxRawPoint*)iter->raw_point)[i].x;
 		int y = ((LivoxRawPoint*)iter->raw_point)[i].y;
 		int z = ((LivoxRawPoint*)iter->raw_point)[i].z;
 		int r = ((LivoxRawPoint*)iter->raw_point)[i].reflectivity;
 		sprintf_s(line, "%d %d %d %d\r\n", x, y, z, r);
 		int len = std::strlen(line);
 		if (curpos + len >= WRITE_BUFFER_LEN)
 		{
 			txt_file.write(buffer.get(), curpos);
 			curpos = 0;
 		}
 		memcpy(buffer.get() + curpos, line, len);
 		curpos += len;
          }
     }
}
[image]

shenshen927, 2020-05-09:
知道啦，是因为datatype不一样，因此数据格式不同。

closed 2020-05-09 by shenshen927

SweetBittErz, 2021-06-11:
我也是这个情况，请问怎么改呢？我用的MID-40的代码读Horizion，出来的图和你这一模一样，请问怎么改才可以变成正确的呢？

Oakks, 2021-07-07:
if(data ->data_type == kExtendCartesian)
        {