Livox-SDK/Livox-SDK #51: False depth values on Object edges

mbcel, 2020-05-14:
I found that on object edges the depth values are extrapolated over the background. This happens where the background behind the edge lies in a shadowed region (the intensity here is zero), however, there, depth values are still given but falsely given by the object contour and not the background. 

Why does this happen? And how can I fix this? It seems to me that this is due to the different position of the emitter and detector of the lidar. But why are the intensity values correct but not the depth values?

mbcel, 2020-05-14:
[image]
(Left intensity, Right depth)
This might better show it. When you directly overlap both images you see that the shadow region is part of the foreground object (forground wall), although in reality it is part of the background.

Livox-SDK, 2020-05-21:
Hi @mbcel . To make it easier to locate the problem. Could you send an email to dev@livoxtech.com with pointcloud's lvx file (use Livox Viewer to record within 20s) and corresponding camera image.

closed 2020-09-07 by Livox-SDK