Livox-SDK/Livox-SDK #53: 如何解决传输数据量过大的问题?

qiziqian, 2020-05-18:
我们的环境中部署了多个mid-100,使用了一个SDK去接收数据,因为传输过来的数据量很大,所以导致服务器负载压力很重,针对这个问题,我有两个方案:
1.减少数据量
2.将设备串行化,轮流接收数据
暂不考虑扩展服务器规模
针对第一个方案,我想要知道可以配置数据量的api
针对第二个方案,我想要知道有没有可以使设备轮流发送数据的api,还是说只能通过start,stop等api自行实现?

Livox-SDK, 2020-05-21:
您好，@qiziqian
1. 针对第一种方案，目前的SDK 都没有可以配置数据量的API.
2. 针对第二种方案，目前SDK也没有使设备轮流发送数据的API，目前只能通过start 和 stop 等API 自行实现.

qiziqian, 2020-05-22:
> 您好，@qiziqian
> 
> 1. 针对第一种方案，目前的SDK 都没有可以配置数据量的API.
> 2. 针对第二种方案，目前SDK也没有使设备轮流发送数据的API，目前只能通过start 和 stop 等API 自行实现.

你好,我还有两个问题:
1.我在翻阅livox api的过程中,发现HubSetImuPushFrequency 这个API,这个是不是意味着我可以通过购买一个livox hub设备来控制数据量?
2.如果我自己实现轮流发送的代码,是不是通过LidarStartSampling  LidarStopSampling 这两个api来启动/停止设备的传输?

Livox-SDK, 2020-05-26:
您好，
1. HubSetImuPushFrequency 是用户设置 IMU 的推送频率的 API，点云的推送频率不支持修改
2. 是的，您可以通过LidarStartSampling LidarStopSampling 这两个api来控制点云传输的开关

closed 2020-09-07 by Livox-SDK