Livox-SDK/Livox-SDK #57: 关于Livox点云的覆盖范围和精度问题

shenshen927, 2020-05-31:
看官网上的宣传说基本能在1秒内接近100%的覆盖率，不知道这个覆盖率是如何定义的？个人感觉应该和视场范围、深度等因素都有联系。
        我目前遇到的问题是这样的，采用Mid-40、Horizon对一个固定的静态的场景进行扫描，如果按照官方标定的1秒左右就可以完成100%覆盖的话，那么我收集5秒的数据后，应该就可以完成背景初始化了，然后再接收实时点云数据，利用深度信息与原始背景进行比较，检测出遗落的其它物体或通过的运动物体。但实际情况是：背景初始化完成后，在实时点云数据与背景进行对比时却发现存在很多深度信息与背景不同的点（但实际情况是完全相同的场景），起初我猜测是不是我背景收集的数据还不够多，于是我把背景初始化的时长增大到30秒（收集30秒的数据，建立不同位置(x,y)的初始深度信息），然而实时对比时仍然存在很多深度信息有明显差异的点。
        我的程序逻辑是对监测区域初始化为一张网格，每个格子的大小是如果是5cm，深度信息对比就是拿实时与背景相对应的网格的深度进行对比。即使如此，对于相同的静态场景（室内），积累30秒背景，实时对比时也仍然有很多深度差异较大的格子。
        请问，这是什么原因？或者我应该如何调整我的程序逻辑？

Livox-SDK, 2020-06-02:
您好，
点云覆盖率的具体定义可以参照官网说明[点云特性](https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/Livox%20%E7%82%B9%E4%BA%91%E7%89%B9%E6%80%A7.pdf) ，计算时视场被划分成 5mrad*5mrad
的⽹格。对于上述提到的问题，点云对比计算时需要考虑深度误差，角度误差等因素，在对比的时候建议考虑一下误差因素。

shenshen927, 2020-06-03:
首先感谢您的解答。还有部分疑问如下：
1、看了点云特性的那张图，那张图是多长时间的扫描积累，是在多远的位置 得到的扫描覆盖率？
我的疑问是：
（1）比如一个1米见方的物体，如果放在距离激光雷达5米远的位置，等待扫描5秒钟，如果能到达90%的覆盖率。那么如果把该物体放到10米远的距离，同样扫描5秒钟，覆盖率是否会下降？（比如变成70%）
（2）如果该10米远的这个物体，增加扫描时长到10秒钟，那么其覆盖率是否也会相应的增加？（比如重新增加到90%）
2、您说的“在对比的时候考虑下误差”能否给个详细的指导，谢谢。
（1）对于深度误差，其误差范围是多少？或者是与距离成正比的关系？
（2）关于角度误差该怎么理解，收到的数据是x、y、z的坐标值，请问角度误差体现在哪里？以及如何消除角度误差影响？

closed 2020-09-07 by Livox-SDK

zzqjh, 2022-11-03:
> 您好， 点云覆盖率的具体定义可以参照官网说明[点云特性](https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/Livox%20%E7%82%B9%E4%BA%91%E7%89%B9%E6%80%A7.pdf) ，计算时视场被划分成 5mrad*5mrad 的⽹格。对于上述提到的问题，点云对比计算时需要考虑深度误差，角度误差等因素，在对比的时候建议考虑一下误差因素。

请问，horizon激光雷达如何调整积分时间并录制数据呢