Livox-SDK/Livox-SDK #60: How often can I get a single UDP data package?

crea397, 2020-06-04:
Hi,

I checked the [Point Cloud {#data_type}](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol#33-point-cloudimu-data-data_type) of the Livox SDK Communication Protocol.
I'm using a Mid-40 and the Data Type is 0.
So I found out that single UDP data package contains 100 samples.

Also, from the documentation and tables
「Taking Mid-40 as an example, the sampling frequency is 100 KHz , and the interval of each point is 10μs」
|Point Num  |Timestamp  |
|---|---|
|1  |T1  |
|2  |T1 + 10 μs  |
|3  |T1 + 20 μs  |
|...  |...  |
|100  |T1 + 990 μs  |

I found the following:
It was found that single UDP data package contains 1000μs (0.001s) worth of data (100 points).
I believe that the Data Frequency, 100,000 points/s, is achieved by collecting 1,000 UDP data packages.

How often can I get a single UDP data package?
I'm thinking data package/0.001s, is this correct?

Thanks!

Livox-SDK, 2020-06-06:
Hi，@crea397 
Yes, you are right!
For Mid-40,  UDP data package frequency is 1000 packages / s.

crea397, 2020-06-06:
Hi, @Livox-SDK 
Thank you for your reply.

I consider, for example, processing a point cloud of data (1s / 100,000 points / 1000 UDP data packages) in real time.

In such a case, is it necessary to complete the process of extracting the point cloud data from the single UDP data package within 0.001s?

Livox-SDK, 2020-06-08:
Yes，it's  it necessary to extracting the point cloud data from the single UDP data package within 0.001s

crea397, 2020-06-08:
I could understand.
I am deeply grateful to you.

closed 2020-06-08 by crea397