Livox-SDK/Livox-SDK #63: Question about the lvx format

kentaroy47, 2020-06-11:
[image]
While the frame header is 32bytes, only the first 24 bytes are specified in the Specifications.
What is the last 8 bytes?

Thanks!
https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/Livox%20Viewer/LVX%20Specifications%20EN_20190924.pdf

kentaroy47, 2020-06-11:
Seems like this is a bug, only 24bytes in the frame header..

Livox-SDK, 2020-06-12:
Hi, @kentaroy47 
Yes, It's a bug on LVX Specifications v1.0. frame header actually only have 24bytes.
Horizon use new version of lvx format V1.1. you can refer to new version of [LVX Specification V1.0](https://terra-1-g.djicdn.com/65c028cd298f4669a7f0e40e50ba1131/Download/LVX%20Specifications.pdf)

kentaroy47, 2020-06-12:
Thanks!
You should link this document to the Horizon download page.. It will cause
much less confusions

closed 2020-07-22 by kentaroy47