Livox-SDK/Livox-SDK #64: apr_build.bat运行报错，环境搭建不了，无法编译

MrJianglc, 2020-06-12:
Livox Team你们好，
我的电脑是WIN10 64位操作系统；CMake版本：3.18.0-rc1-win64-x64；VS版本：Visual Studio 2017 Professional。操作如下
1、CMake生成工程（界面中选择输入输出路径后，Configure下选择自己的vs2017版本以及操作系统版本，完成后再点击Generate，生成成功）
2、apr文件夹下右键打开powershell,输入“.\apr_build.bat amd64”回车。出现了如下报错
PS D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr> .\apr_build.bat amd64

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>Copyright (c) 2019 Livox. All rights reserved.
'Copyright' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>Permission is hereby granted, free of charge, to any person obtaining a copy
'Permission' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>of this software and associated documentation files (the "Software"), to deal
'of' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>in the Software without restriction, including without limitation the rights
'in' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
'to' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>copies of the Software, and to permit persons to whom the Software is
'copies' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>furnished to do so, subject to the following conditions:
'furnished' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>The above copyright notice and this permission notice shall be included in
'The' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>all copies or substantial portions of the Software.
'all' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
'THE' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
'IMPLIED' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
'FITNESS' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
'AUTHORS' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
'LIABILITY' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
'OUT' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>SOFTWARE.
'SOFTWARE.' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

D:\DJ\Livox-SDK-master\Livox-SDK-master\third_party\apr>echo off
nmake: *** No targets specified and no makefile found.  Stop.
nmake: *** No rule to make target `install'.  Stop.
系统找不到指定的文件。

lonlonago, 2020-06-12:
same problem @karolmajek

Livox-SDK, 2020-06-12:
@lonlonago @MrJianglc ,
建议您可以先使用 git clone 的方式下载源码。
目前，以压缩包下载的方式会有该问题，我们会尽快修复！

MrJianglc, 2020-06-14:
@Livox-SDK 非常感谢！果然是这个问题，问题得到解决了。也怪我没仔细阅读文档（文档中有提到使用git clone的方式）。

Livox-SDK, 2020-06-17:
APR 在 Windows 上的编译错误已经在 V2.1.1 发行版本后解决 （当源码是通过 ZIP 包下载时）

APR Compiled error on Windows has been fix since Release V2.1.1 （when source code is download from ZIP )

closed 2020-06-30 by MrJianglc