Livox-SDK/Livox-SDK #66: Rays not returning don't have Azimuth and Zenith angles

tgurriet, 2020-06-14:
Hi Livox team,

Is their a way to get the Azimuth and Zenith angles for the rays that do not return?
Right know is seems that all the fields are 0.

Best,
Tom

Livox-SDK, 2020-06-23:
Hi @tgurriet ,
You can use this API
 ` livox_status SetSphericalCoordinate ( uint8_t handle, CommonCommandCallback cb,  void *  client_data )`
to change point cloud coordinate system to spherical coordinate. And then you can get the  Azimuth and Zenith angles of point clouds that do not return.

tgurriet, 2020-06-23:
I do use the SetSphericalCoordinate command, but all points that have depth=0. (which I assume are the points that do not return) also have their Azimuth and Zenith angles at 0.
Also that's with a Mid-100 and firemware LIVOX_MID_FW_03.02.

Livox-SDK, 2020-06-24:
Hi @tgurriet ,
This version of firmware not support to get the Azimuth and Zenith angles that do not return.
It's recommended to upgrade your Mid-100 to latest firmware version by Livox Viewer.
[LIVOX_MID_FW_03.07.0000.bin](https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/20200107/LIVOX_MID_FW_03.07.0000.bin)

tgurriet, 2020-06-24:
@Livox-SDK 
I've updated the firmware of the MID-100 and the one of the HUB, but I still get a lot of points that return with all their fields to 0 in spherical mode.

Livox-SDK, 2020-06-29:
Hi @tgurriet 
We are sorry that this is a bug on Hub to process points in spherical mode. And we will fix this later.

tgurriet, 2020-06-29:
Thank you.

Do you have a rough idea of when this fix will be out so we can adapt our development schedule (days, weeks, months)?

Livox-SDK, 2020-07-02:
Hi @tgurriet
It's recommended to contact our support team directly cs@livoxtech.com who will provide you with a more detailed plan.

closed 2020-09-07 by Livox-SDK