Livox-SDK/Livox-SDK #69: Parsing .lvx Files

HungrySkeleton, 2020-06-19:
Hello Livox Team,

I was taking a look at the .lvx specification sheet and I was wondering if there is any easy way to parse the data stored in the lvx frame rather than extraction all frames directly from the binary file? Is there something in the will read an lvx file and allow us to just access frames directly? Or do we have to individually extract the frames since we know the format, format duration and size of the bytes of each frame. I have looked around in the API and there doesn't seem to be any read functions there is only a write function which saves frames.

Livox-SDK, 2020-06-23:
Hi @HungrySkeleton ,
Currently, there is no API to read an lvx file.  But this [ example of reading lvx file](https://github.com/Livox-SDK/livox_ros_driver/blob/master/livox_ros_driver/livox_ros_driver/lds_lvx.cpp) might help you.

xmba15, 2020-07-11:
@HungrySkeleton @Livox-SDK 
I created a library to help manipulate lvx files, hope this can help.

HungrySkeleton, 2020-07-13:
Hey thanks a lot @xmba15

closed 2020-09-07 by Livox-SDK