Livox-SDK/Livox-SDK #73: UDP Broadcast not working in secured network. Is UDP Unicast possible?

codegrafix, 2020-07-09:
Due to some network regulations broadcast is not allowed in our network.
Is it possible to enable UDP unicast?

For example with Velodyne sensors its possible to specify the destination port and ip address so that not all other devices get flooded.
Can you please make this also available?

We plan to run multiple sensors in our network and we would highly welcome a fast reply on this.

Livox-SDK, 2020-07-10:
Hi, @codegrafix 

Currently, It's not possible to use UDP unicast only.
Handshake mechanism between LiDAR and Livox-SDK relys on UDP broadcast.
But after connection establishment，LiDAR will communicate with Livox-SDK with UDP unicast.

codegrafix, 2020-07-15:
Hi @Livox-SDK,
thanks for the information. Do you have any plans to support a direct connection without UDP for the future?

Livox-SDK, 2020-07-20:
Hi @codegrafix 
We still have no plans to do it at present.

codegrafix, 2020-07-21:
O.K. I'm looking forward to this!

codegrafix, 2020-09-02:
Is it possible to skip the broadcast/handshake in order to create a direct connection? **Currently this is a show stopper!** Would be very happy If you can provide at least some recommendations.

closed 2020-09-07 by Livox-SDK