Livox-SDK/Livox-SDK #75: livox SDK for Python

roya-correct-ai, 2020-07-11:
Hi there

Could you please let me know how I can capture the lidar data in Python? since in Livox website (https://www.livoxtech.com/horizon), it is mentioned that "Livox SDK supports development languages, including C, C++, and Python on multiple platforms" but I could not find any Python library in your Github.

I appreciate it if you help me with this issue
Thanks in advance
Roya

Livox-SDK, 2020-07-14:
Hi @roya-correct-ai,
There is a third-party python driver available
https://github.com/ryan-brazeal-ufl/OpenPyLivox

roya-correct-ai, 2020-07-14:
> Hi @roya-correct-ai,
> There is a third-party python driver available
> https://github.com/ryan-brazeal-ufl/OpenPyLivox

Thank you for your response.
I have seen that library, It works well to capture the data but, it has not equipped with real-time processing until now, I can capture the data but I can not process the data unless the capturing finishes and this procedure is not real-time.
I want to have a process on each sample then work on the next one.
I appreciate it if you help me.

Livox-SDK, 2020-07-20:
Hi @roya-correct-ai 
We are sorry that there is no livox python library that can process data in real time until now.

Vldub, 2020-08-06:
[image]
wtf

Vldub, 2020-08-06:
![sad](https://user-images.githubusercontent.com/35417317/89511969-3da11d00-d7db-11ea-8a1f-dd0f0506a14c.png
why do you write that there is support?

mar2194, 2020-09-15:
If you are trying to read .lvx files using Python, there is now a .lvx file reader as part of the OpenPyLivox library.
https://github.com/ryan-brazeal-ufl/OpenPyLivox

siyangxie, 2021-11-24:
> https://user-images.githubusercontent.com/35417317/89511969-3da11d00-d7db-11ea-8a1f-dd0f0506a14c.png why do you write that there is support?

@Livox-SDK Please advise.