Livox-SDK/Livox-SDK #76: How to capture IMU data from Horizon?

roya-correct-ai, 2020-07-17:
Hello 

I want to capture the lidar IMU data. As you mentioned in the communication protocol, the IMU data can be captured if the data type is equal to 6. However, when I read the lidar package's data-type, it sent me number-2 which means the cartesian XYZ.
Could you please guide me on how I can request from the lidar to send me the package with data type=6 (IMU data)?
Should I force the lidar to change the package's data-type? I appreciate it if you tell me how it is possible to capture IMU data.

Many thanks in advance
Roya

Livox-SDK, 2020-07-20:
Hi @roya-correct-ai 
Horizon or Tele ‘s IMU data will be pushed at a frequency of 200Hz by defalut.  And It's mixed with point cloud data in same data callback.
If data type equals to 6, it indicates IMU data, if data type equals to 2, it indicates point cloud data.
If  you don't recevie any data type equals to 6 all the time. Please use Livox Viewer checking LiDAR's setting (whether IMU push is enabled).

roya-correct-ai, 2020-07-29:
> Hi @roya-correct-ai
> Horizon or Tele ‘s IMU data will be pushed at a frequency of 200Hz by defalut. And It's mixed with point cloud data in same data callback.
> If data type equals to 6, it indicates IMU data, if data type equals to 2, it indicates point cloud data.
> If you don't receive any data type equals to 6 all the time. Please use Livox Viewer checking LiDAR's setting (whether IMU push is enabled).

Thank you for your response.
I checked the IMU push frequency in Livox viewer and it is set on 200 MHz but, unfortunately, when I read the data_type from lidar packet data, It is only equal to 2 and there is no data_type=6 at all. 
How I can be sure that My LiDAR (Livox Horizon) is equipped by the IMU sensor? 
Could you please guide me on what should I do to enable and capture the IMU data?

Many thanks in advance
Roya

Pirkaklo, 2021-04-16:
Dear roya, I do not know if you solved this problem. But I want to show one of the solving ways to everyone who face this problem. 
    If you set the IMU frequency as 200 Hz, it can be confirmed that the IMU function is opened. Then if you want to turn lvx files to rosbag files, please ensure that the "arg name="imu_bag" default" is true(the default setting is false); in the  lvx_to_rosbag.launch, the path of this file usually in this way 
    ~/ws_livox/src/livox_ros_driver/launch/lvx_to_rosbag.launch. 
    Then you can get the IMU data after you finish the converting.

Pirkaklo, 2021-04-16:
@roya-correct-ai 
    Dear roya, I do not know if you solved this problem. But I want to show one of the solving ways to everyone who face this problem. 
    If you set the IMU frequency as 200 Hz, it can be confirmed that the IMU function is opened. Then if you want to turn lvx files to rosbag files, please ensure that the "arg name="imu_bag" default" is true(the default setting is false); in the  lvx_to_rosbag.launch, the path of this file usually in this way 
    ~/ws_livox/src/livox_ros_driver/launch/lvx_to_rosbag.launch. 
    Then you can get the IMU data after you finish the converting.

Pirkaklo, 2021-04-16:
@roya-correct-ai 
    Dear roya, I do not know if you solved this problem. But I want to show one of the solving ways to everyone who face this problem. 
    If you set the IMU frequency as 200 Hz, it can be confirmed that the IMU function is opened. Then if you want to turn lvx files to rosbag files, please ensure that the "arg name="imu_bag" default" is true(the default setting is false); 
in the lvx_to_rosbag.launch, the path of this file usually in this way :
    ~/ws_livox/src/livox_ros_driver/launch/lvx_to_rosbag.launch. 
   Then you can get the IMU data after you finish the converting.

Pirkaklo, 2021-04-16:
Dear roya, I do not know if you solved this problem. But I want to show one of the solving ways to everyone who face this problem. 
    If you set the IMU frequency as 200 Hz, it can be confirmed that the IMU function is opened. Then if you want to turn lvx files to rosbag files, please ensure that the "arg name="imu_bag" default" is true(the default setting is false); in the  lvx_to_rosbag.launch, the path of this file usually in this way 
    ~/ws_livox/src/livox_ros_driver/launch/lvx_to_rosbag.launch. 
    Then you can get the IMU data after you finish the converting.
@roya-correct-ai

Pirkaklo, 2021-04-16:
@roya-correct-ai 
Dear roya, I do not know if you solved this problem. And I want to show one of the solving ways to everyone who face this problem. 
If you set the IMU frequency as 200 Hz, it can be confirmed that the IMU function is opened. Then if you want to turn lvx files to rosbag files, please ensure that the "arg name="imu_bag" default" is true(the default setting is false); in the  lvx_to_rosbag.launch, the path of this file usually in this way 
    ~/ws_livox/src/livox_ros_driver/launch/lvx_to_rosbag.launch. 
Then you can get the IMU data after you finish the converting.

panmengjia, 2021-05-07:
Is this problem solved? I don’t know how to get imu data when using avia.

TBRTK, 2021-12-05:
> Is this problem solved? I don’t know how to get imu data when using avia.

Bumping this one. 
Is IMU accessible for the AVIA ?

WingmanMedia, 2022-02-21:
I know it is all about SDK here but can I ask a question about a point cloud captured through Livox Viewer with Horizon/Avia/Tele-15. 

I have just downloaded a sample data from Horizon which comes with IMU on board but all O x,y,z fields have no values in them.  Does it mean that if you use Livox Viewer to capture a point cloud with Horizon/Avia/Tele-15 that come with IMU and you turn it on in the lidar settings you still won't get it recorded by Livox Viewer.

rlabs-oss, 2022-05-04:
What "float" format is the float for the IMU, specifically does the SDK include any function to convert it? Ive tried dumping the bytes to float memory but that doesnt work. 

Points 1537,-543,47,103,16,481899510060
IMU 146229248.000000,997922880.000000,998836544.000000,1065247296.000000,3149374208.000000,977610816.000000,481899234690
Points 0,0,0,0,0,481899907890

uint32_t was easy enough for the points, but just stating "float" in the SDK without any example code extracting / using the data is really confusing.