Livox-SDK/Livox-SDK #78: 如何使用SDK进行UTC时间同步

sandy0209, 2020-07-30:
技术您好，对于UTC时间同步有如下几个问题：
1.使用SDK中的lidar_utc_sync测试发现，控制台会不断输出”[WARING] Lidar sn: [1HDDGC500102301] StateChange!!!”这条语句，为什么会不断调用OnDeviceChange回调函数，运行该过程是当type=kEventStateChange，含义为工作状态改变或有错误发生，难道是同步过程中Lidar的工作状态会不断变化？还是说发生了错误？
2.如何utc同步来进行采样保存lvx文件呢？目前参考lidar_utc_sync与lidar_lvx_sample结合，控制逻辑是连接Lidar并配置同步，使用LidarStartSampling来进行采样，但收到的数据包时间戳并不是UTC时间。该如何在采样程序中嵌入同步UTC？目前不知如何进行使用

sandy0209, 2020-07-30:
使用的是Horizon激光雷达

Grandit, 2020-08-05:
同问，另，如果采用IEEE1588-2008时间同步，有samplecode吗？

Livox-SDK, 2020-08-07:
您好 @sandy0209，
1. 推送 kEventStateChange 有两种情况，一种是设备初始化状态，还有一种就是设备推送警告信息（脏污预警等）。
2. utc 同步的同时采样保存lvx 文件，需要结合lidar_utc_sync与lidar_lvx_sample。收到的数据包不是UTC时间，建议您从两方面确认一下：1. 是否成功读取和解析GPRMC数据；2. 是否成功发送UTC 时间同步命令给LiDAR( 在LdsLidar::SetRmcSyncTime 函数中打印日志确认)

closed 2020-09-07 by Livox-SDK