Livox-SDK/Livox-SDK #79: [Horizon] how to sync. timestamp?

Grandit, 2020-08-05:
Hello,  I have two questions:
1.  Livox Viewer:
-- Can Livox Viewer support IEEE1588-2008 timestamp sync. when connects to Horizon? If so, how to set it?
2.  sample code：
-- How to use IEEE1588 to sync. timestamp? 
-- And just send&receive ptp packet(sync，follow_up, delay_req,delay_resp) following with the IEEE1588 protocol? Is there anything should pay attention to?

Livox-SDK, 2020-08-07:
Hi @Grandit，
[Timestamp Synchronization](https://github.com/Livox-SDK/Livox-SDK/wiki/Timestamp-Synchronization) might help you.

Grandit, 2020-08-07:
Thanks for your answer!
It means that I _can't sync._ the timestamp by PTP protocol **unless** my eth. hardware support PTP hardware clock?
_And my eth. hardware infomation as follows:_
[image]

Grandit, 2020-08-11:
Hi，
    I have tested the PPS timestamp sync. type,  but I **couldn't** find  the expection that "**_the rising edge of PPS will reset the timestamp to zero_**".
    And any other procedure did I miss? 
    Please help~ thx!

smallzhao, 2020-10-11:
I also encountered the same problem. My network card does not support hard clocks. Can we support soft clocks?