Livox-SDK/Livox-SDK #90: Save the data with the GPS synchronization

meng-zha, 2020-09-30:
Thanks for your code of time synchronization.
However, I'm still a little confused of the way to save pointclouds with GPS sync.
Should I modify the timestamp of every packet after running the time sync? And I want to know where I should add the code
to save the lvx data as well as sync code.
Hope there will be the code of data acquisition as well as time sync together.

meng-zha, 2020-10-01:
I have tried to combine the [lidar_utc_sync code](https://github.com/Livox-SDK/Livox-SDK/tree/master/sample_cc/lidar_utc_sync) with the [lidar_lvx_file code](https://github.com/Livox-SDK/Livox-SDK/tree/master/sample/lidar_lvx_file), but I find that the lvx file's timestamp type is still **0**, which means **No sync source**. Thanks for the any advice.

meng-zha, 2020-10-02:
The problem is solved by waiting for two seconds before sampling the point after synchronization.

closed 2020-10-02 by meng-zha

supercrazysam, 2023-10-31:
> The problem is solved by waiting for two seconds before sampling the point after synchronization.

Hi I am trying to setup GPS sync for livox lidar as well.  I supplied PPS pulse and sent GPRMC,   the lidar timestamp_type  becomes 4 (PPS) and stuck there.   It never gets to timestamp_type 3 (GPS) as described in the livox timesync manual.

Were you able to get timestamp_type to turn into 3? @meng-zha

tranhien1612, 2025-07-05:
@meng-zha Can you share your code with me?