Livox-SDK/Livox-SDK #93: Time Synchronization of Mid-100/Mid-40 using PTP

AvGladiss, 2020-10-22:
I am trying to time-sync my Mid-100 (Firmware version 3.8) to a PTP master clock (hardware clock available and used). I have set up the PTP master clock following your instructions in https://github.com/Livox-SDK/Livox-SDK/wiki/Timestamp-Synchronization#Appendix. Using Wireshark, we see `Sync Messages` and `Follow Up` messages from the PTP master clock, and `Delay_Req messages` from the Mid40s. However, Livox Viewer 0.8.0 states that there is no 1588 Signal.

AvGladiss, 2020-10-22:
I am sorry, I forgot to open UDP port 319 for receiving the `Delay_Req` messages and therefore, the grand master did not send the `Delay_Resp` messages....

closed 2020-10-22 by AvGladiss

IKAROS93, 2020-11-24:
HI，my friend ,could you please tell me how to open the UDP port 319(139)? THANKS!@AvGladiss