Livox-SDK/Livox-SDK #94: How can we turn on and off the lidar using the sdk?

AJMena, 2020-11-04:
Hello there.
We currently have a Livox Horizon installed in a factory. The lidar is used to scan some of the trucks that enter the factory. So the Lidar is working mostly all day, seven days a week. We want to know how we can turn on and off the lidar so the scanner is on only when is necessary. That way to longer the lidar's lifespan.
We hope that there is a way to do that using the livox sdk. If you can point us in the good direction would help us a lot.
Thank you.

Livox-SDK, 2020-11-09:
Hi @AJMena 
Lidar can be switch between Power-Saving Mode and Normal Mode. 
In Power-Saving Mode, scanner and motor will stop work.
You can use this API to change working mode [LidarSetMode](https://livox-sdk.github.io/Livox-SDK-Doc/livox__sdk_8h.html#af5e112560f926a14b9b37e03f76b5a6b).

closed 2020-12-15 by Livox-SDK

trishmapow, 2021-12-01:
I hate to re-open this issue, but my Livox Horizon seems to restart automatically after I put it in power saving mode. This happens also using the Viewer software. As soon as I put it in power saving, the fan stops, but 10 seconds later it starts up again. Would appreciate a response because it is critical that the lidar can be switched off. Thanks.

gabrielheider, 2022-05-05:
Hi, did someone solve this problem? Can someone provide a example how to change working mode in c++ code with the SDK?