Livox-SDK/Livox-SDK #95: LOG_ERROR("DeviceDiscovery Create Socket Failed")

cuiDarchan, 2020-11-06:
in device_discovery.cpp  file
in  bool DeviceDiscovery::Start  function
SDK2.0 version . 
In most cases, the device link is normal. 
Occasionally LOG_ERROR("DeviceDiscovery Create Socket Failed") .
There are 7 mid40 connected to the network .
Thank you very much !

cuiDarchan, 2020-11-06:
Env: Ubuntu 16.04

Livox-SDK, 2020-11-09:
Hi @cuiDarchan 
This is because SDK2.0 doesn‘t create socket with reuse_port option.
This bug has been fix. It's recommended to update your SDK version to 2.2.0.

cuiDarchan, 2020-11-16:
> Hi @cuiDarchan
> This is because SDK2.0 doesn‘t create socket with reuse_port option.
> This bug has been fix. It's recommended to update your SDK version to 2.2.0.
Thank you very much！

closed 2020-11-16 by cuiDarchan