Livox-SDK/Livox-SDK #97: Lidar CMD 0x06 support for Avia triple return mode

ryan-brazeal-ufl, 2020-11-06:
Hello,
Can you please update the SDK to include how to command an Avia sensor to utilize triple return mode. Within the [current SDK document for lidar command 0x06: set lidar return mode](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol#0x06-set-lidar-return-mode) only single and dual return modes are listed.
Thanks,
Ryan

Livox-SDK, 2020-11-09:
Hi @ryan-brazeal-ufl 
Thank you for pointing this out. 
SDK docunment has been updated. [triple mode correseponding to 0x03.](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol#0x06-set-lidar-return-mode)

ryan-brazeal-ufl, 2020-11-16:
I believe you also need to update **0x07 Get LiDAR Return Mode** documentation with the 0x03 return value.

rlabs-oss, 2020-11-26:
Are there any plans to update the Mid-40 firmware to support multiple returns in one firmware version?

Livox-SDK, 2020-12-16:
> I believe you also need to update **0x07 Get LiDAR Return Mode** documentation with the 0x03 return value.
**0x07 Get LiDAR Return Mode** also has been updated.

Livox-SDK, 2020-12-16:
> Are there any plans to update the Mid-40 firmware to support single and multiple returns in one firmware version?
Hi @rlabs-oss 
We are sorry that we have no plans to update the Mid40 firmware version so far.

closed 2020-12-16 by Livox-SDK