Livox-SDK/Livox-SDK #98: Livox Hub 在配置好ptp主时钟后无法进行同步

gloryhry, 2020-11-14:
我们的Livox Hub通过连接思科的交换机与5095工控机进行网络链接，Livox Hub使用的动态Ip，Ip由华为的4G路由器进行分配。
在5095工控机上根据[教程](https://github.com/Livox-SDK/Livox-SDK/wiki/livox-device-time-synchronization-manual-cn#%E9%99%84%E5%BD%95)设置好了PTP主时钟，且在wireshark中能看到Sync、Follow_Up这两个数据。

但是进入Livox viewer中Livox Hub并没有显示1588 sync。且输入`sudo phc2sys -c eth0 -s CLOCK_REALTIME -O 0`这行代码后终端无输出，一直保持运行。

是我哪里设置的有问题吗？请尽快回复

gloryhry, 2020-11-14:
[image]

IKAROS93, 2020-11-24:
请问这个问题解决了吗？我的horizon也是遇到了同样的问题。

gloryhry, 2020-11-24:
我最后通过配置ptpd服务可对livox hub进行同步，但是开启ptpd服务不知为何会使同一网关中的路由器死机，因此最后我通过修改ros驱动来尽量减小时间差。
我虽然通过邮件联系过livox的官方，但是回复邮件仅问我是否按照官方文档进行配置，没有任何正面回复我提出的问题。

IKAROS93, 2020-11-25:
非常感谢，在采用您的建议后，我们使用ptpd成功进行了时间同步。
[参考链接](https://blog.csdn.net/u013431916/article/details/83054369?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522160626950919725266964614%2522%252C%2522scm%2522%253A%252220140713.130102334..%2522%257D&request_id=160626950919725266964614&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~all~top_click~default-1-83054369.first_rank_ecpm_v3_pc_rank_v2&utm_term=ptpd&spm=1018.2118.3001.4449)

closed 2020-12-15 by Livox-SDK

FengYuQ, 2023-05-23:
@gloryhry 方便给个联系方式吗？最近用思科的交换机做同步一直不能成功

gloryhry, 2023-05-24:
可以参考 @IKAROS93 给的链接进行ptpd进配置。或许交换机没开放相关端口？或者不支持ptp协议？

我觉得可以试试先换一个交换机看看

FengYuQ, 2023-05-27:
> 

交换机需要开放哪些端口呢？

FengYuQ, 2023-05-27:
@gloryhry 想问下交换机需要开放哪几个相关的端口呢？