Livox-SDK/Livox-SDK #99: Using the LivoxRawPoint data?

rlabs-oss, 2020-11-14:
In C could you show how to output the points (e.g printf) based on the structures:

/** Cartesian coordinate format. */
typedef struct {
  int32_t x;            /**< X axis, Unit:mm */
  int32_t y;            /**< Y axis, Unit:mm */
  int32_t z;            /**< Z axis, Unit:mm */
  uint8_t reflectivity; /**< Reflectivity */
} LivoxRawPoint;

The ethpacket is 

/** Point cloud packet. */
typedef struct {
  uint8_t version;              /**< Packet protocol version. */
  uint8_t slot;                 /**< Slot number used for connecting LiDAR. */
  uint8_t id;                   /**< LiDAR id. */
  uint8_t rsvd;                 /**< Reserved. */
  uint32_t err_code;      /**< Device error status indicator information. */
  uint8_t timestamp_type;       /**< Timestamp type. */
  /** Point cloud coordinate format, refer to \ref PointDataType . */
  uint8_t data_type;
  uint8_t timestamp[8];         /**< Nanosecond or UTC format timestamp. */
  uint8_t data[1];              /**< Point cloud data. */
} LivoxEthPacket;

As points arrive in real time I would like to dump them to STDOUT as ints so they can be piped and consumed by other processes.

The getlidardata function in the c examples is obfuscated and not at all clear to understand hence the request. 

Thank you.

closed 2020-11-14 by rlabs-oss

ZhixiongSun, 2021-04-15:
Hey @rlabs-oss ,
Have you solved this problem, can you tell me how you did it, thank you

rlabs-oss, 2021-04-15:
modify the sample_cc/lidar/lds_lidar.cpp example.
At the end of the if(eth_packet) simply print it

` }
    // only valid for kCartesian, DUMP XYZI to stdout, mm 
    for(int i=0;i<100*sizeof(LivoxRawPoint);i+=13) {   // XXXXYYYYZZZZI
    printf( "%d,%d,%d,%d,%ld\n", ((int32_t) ((data->data[i]) + (data->data[i+1]<<8) + (data->data[i+2]<<16) + (data->data[i+3]<<24))) , 
                             ((int32_t) ((data->data[i+4]) + (data->data[i+5]<<8) + (data->data[i+6]<<16) + (data->data[i+7]<<24))) , 
                             ((int32_t) ((data->data[i+8]) + (data->data[i+9]<<8) + (data->data[i+10]<<16) + (data->data[i+11]<<24))), 
                             (uint8_t) data->data[i+12],(*((uint64_t *)(data->timestamp)))+i);
    }

`
Change all printf statements in the whole SDK to fprintf(stderr,".. otherwise stdout will contain garbage from the various status messages.

ZhixiongSun, 2021-04-15:
Thank you for your response, with this method all points can be printed out, but do you know how can they be divided into point cloud frame. e.g. 100ms

rlabs-oss, 2021-04-15:
That's determined by the application consuming/reading the stream of points. If the sdk is modified as above it will dump 100 points at a time from the Ethernet frame - which is continuous. If you read 10000 points (or lines) in this case it corresponds to 100ms for the Mid 40. Typically 3000 points or 30ms is good for LOAM and derivatives.

ZhixiongSun, 2021-04-16:
Hi @rlabs-oss , Thanks again for your information, it is very helpful.