Livox-SDK/Livox-SDK #102: Device State Error Code 0X00000000

Ganymedes-wang, 2020-12-16:
执行` ./lidar_sample`，
显示“StateChange”和“Device State Error Code”，输出日志文件如下：

`Livox SDK initializing.
Livox SDK has been initialized.
Livox SDK version 2.3.0 .
Start discovering device.

[2020-12-16 15:37:25.866] [console] [info]  Broadcast broadcast code: 0TFDH9U006G7851  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 0TFDH9U006G7851
[2020-12-16 15:37:25.866] [console] [info] LocalIP: 192.168.1.39  [device_discovery.cpp] [OnBroadcast] [201]
[2020-12-16 15:37:25.866] [console] [info] DeviceIP: 192.168.1.38  [device_discovery.cpp] [OnBroadcast] [202]
[2020-12-16 15:37:25.866] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [203]
[2020-12-16 15:37:25.866] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [204]
[2020-12-16 15:37:25.890] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
[2020-12-16 15:37:25.890] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [100]
[2020-12-16 15:37:25.890] [console] [info] Broadcast Code: 0TFDH9U006G7851  [device_discovery.cpp] [OnData] [101]
[2020-12-16 15:37:25.890] [console] [info] Type: 1  [device_discovery.cpp] [OnData] [102]
[2020-12-16 15:37:25.890] [console] [info] IP: 192.168.1.38  [device_discovery.cpp] [OnData] [103]
[2020-12-16 15:37:25.890] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [104]
[2020-12-16 15:37:25.890] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [105]
[2020-12-16 15:37:25.890] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [243]
[2020-12-16 15:37:25.914] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [118]
[WARNING] Lidar sn: [0TFDH9U006G7851] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2020-12-16 15:37:25.914] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [243]
[2020-12-16 15:37:25.938] [console] [info]  Update State to 1, device connect true  [device_manager.cpp] [UpdateDeviceState] [257]
[WARNING] Lidar sn: [0TFDH9U006G7851] StateChange!!!
Device Working State 1
Device State Error Code 0X00000000
Device feature 0
[2020-12-16 15:37:25.938] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [118]
firm ver: 3.8.0.0
[2020-12-16 15:37:25.938] [console] [info]  Send Command: Set 0 Id 4 Seq 6  [command_channel.cpp] [Send] [243]
[2020-12-16 15:37:25.962] [console] [info]  Recieve Ack: Set 0 Id 4 Seq 6  [command_handler.cpp] [OnCommand] [118]
OnSampleCallback statue 0 handle 0 response 0 
data_type 0 packet num 100
data_type 0 packet num 200
data_type 0 packet num 300
data_type 0 packet num 400
data_type 0 packet num 500
data_type 0 packet num 600
data_type 0 packet num 700
data_type 0 packet num 800
data_type 0 packet num 900
data_type 0 packet num 1000
data_type 0 packet num 1100
data_type 0 packet num 1200
data_type 0 packet num 1300
data_type 0 packet num 1400
data_type 0 packet num 1500
data_type 0 packet num 1600
data_type 0 packet num 1700
data_type 0 packet num 1800
data_type 0 packet num 1900
data_type 0 packet num 2000
data_type 0 packet num 2100
data_type 0 packet num 2200
data_type 0 packet num 2300
data_type 0 packet num 2400
data_type 0 packet num 2500
data_type 0 packet num 2600
data_type 0 packet num 2700
data_type 0 packet num 2800
data_type 0 packet num 2900
data_type 0 packet num 3000
data_type 0 packet num 3100
data_type 0 packet num 3200
[... 238 lines cut ...]
data_type 0 packet num 27100
data_type 0 packet num 27200
data_type 0 packet num 27300
data_type 0 packet num 27400
data_type 0 packet num 27500
data_type 0 packet num 27600
data_type 0 packet num 27700
data_type 0 packet num 27800
data_type 0 packet num 27900
data_type 0 packet num 28000
data_type 0 packet num 28100
data_type 0 packet num 28200
data_type 0 packet num 28300
data_type 0 packet num 28400
data_type 0 packet num 28500
data_type 0 packet num 28600
data_type 0 packet num 28700
data_type 0 packet num 28800
data_type 0 packet num 28900
data_type 0 packet num 29000
data_type 0 packet num 29100
data_type 0 packet num 29200
data_type 0 packet num 29300
[2020-12-16 15:37:55.324] [console] [info]  Send Command: Set 0 Id 4 Seq 42  [command_channel.cpp] [Send] [243]`
请问这是什么原因，谢谢大家！

Livox-SDK, 2020-12-18:
@Ganymedes-wang 
您好，这是正常的日志输出

Ganymedes-wang, 2020-12-18:
@Livox-SDK 您好，那这个日志输出结果存放在什么位置，如何查看数据？

closed 2020-12-21 by Ganymedes-wang