Livox-SDK/Livox-SDK #103: I can not receive data from Livox

iSIlya, 2020-12-20:
Ubuntu 16.04, Livox Horizon (2 pcs.)
Console log:

./lidar_lvx_sample 
Livox SDK initializing.
Livox SDK has been initialized.
Livox SDK version 2.3.0 .
Start discovering device.
[2020-12-20 16:07:26.022] [console] [info]  Broadcast broadcast code: 1HDDH1200100451  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 1HDDH1200100451
[2020-12-20 16:07:26.022] [console] [info] Broadcast code : 1HDDH1200100451 not add to connect  [device_discovery.cpp] [OnBroadcast] [163]
[2020-12-20 16:07:26.024] [console] [info]  Broadcast broadcast code: 1HDDH3200100471  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 1HDDH3200100471
[2020-12-20 16:07:26.024] [console] [info] Broadcast code : 1HDDH3200100471 not add to connect  [device_discovery.cpp] [OnBroadcast] [163]
[2020-12-20 16:07:27.030] [console] [info]  Broadcast broadcast code: 1HDDH1200100451  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 1HDDH1200100451
[2020-12-20 16:07:27.030] [console] [info] Broadcast code : 1HDDH1200100451 not add to connect  [device_discovery.cpp] [OnBroadcast] [163]
[2020-12-20 16:07:27.031] [console] [info]  Broadcast broadcast code: 1HDDH3200100471  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 1HDDH3200100471
[2020-12-20 16:07:27.031] [console] [info] Broadcast code : 1HDDH3200100471 not add to connect  [device_discovery.cpp] [OnBroadcast] [163]
[2020-12-20 16:07:28.038] [console] [info]  Broadcast broadcast code: 1HDDH1200100451  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 1HDDH1200100451
[2020-12-20 16:07:28.039] [console] [info] LocalIP: 192.168.1.209  [device_discovery.cpp] [OnBroadcast] [201]
[2020-12-20 16:07:28.039] [console] [info] DeviceIP: 192.168.1.211  [device_discovery.cpp] [OnBroadcast] [202]
[2020-12-20 16:07:28.039] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [203]
[2020-12-20 16:07:28.039] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [204]
[2020-12-20 16:07:28.039] [console] [info]  Broadcast broadcast code: 1HDDH3200100471  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 1HDDH3200100471
[2020-12-20 16:07:28.040] [console] [info] LocalIP: 192.168.1.209  [device_discovery.cpp] [OnBroadcast] [201]
[2020-12-20 16:07:28.040] [console] [info] DeviceIP: 192.168.1.212  [device_discovery.cpp] [OnBroadcast] [202]
[2020-12-20 16:07:28.040] [console] [info] Command Port: 55502  [device_discovery.cpp] [OnBroadcast] [203]
[2020-12-20 16:07:28.040] [console] [info] Data Port: 56002  [device_discovery.cpp] [OnBroadcast] [204]
[2020-12-20 16:07:28.062] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
[2020-12-20 16:07:28.062] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [100]
[2020-12-20 16:07:28.062] [console] [info] Broadcast Code: 1HDDH1200100451  [device_discovery.cpp] [OnData] [101]
[2020-12-20 16:07:28.062] [console] [info] Type: 3  [device_discovery.cpp] [OnData] [102]
[2020-12-20 16:07:28.062] [console] [info] IP: 192.168.1.211  [device_discovery.cpp] [OnData] [103]
[2020-12-20 16:07:28.062] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [104]
[2020-12-20 16:07:28.062] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [105]
[2020-12-20 16:07:28.063] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [243]
[2020-12-20 16:07:28.064] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
[2020-12-20 16:07:28.064] [console] [info] Handle: 1  [device_discovery.cpp] [OnData] [100]
[2020-12-20 16:07:28.064] [console] [info] Broadcast Code: 1HDDH3200100471  [device_discovery.cpp] [OnData] [101]
[2020-12-20 16:07:28.064] [console] [info] Type: 3  [device_discovery.cpp] [OnData] [102]
[2020-12-20 16:07:28.064] [console] [info] IP: 192.168.1.212  [device_discovery.cpp] [OnData] [103]
[2020-12-20 16:07:28.064] [console] [info] Command Port: 55502  [device_discovery.cpp] [OnData] [104]
[2020-12-20 16:07:28.064] [console] [info] Data Port: 56002  [device_discovery.cpp] [OnData] [105]
[2020-12-20 16:07:28.064] [console] [info]  Send Command: Set 0 Id 2 Seq 5  [command_channel.cpp] [Send] [243]
[2020-12-20 16:07:28.086] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [118]
OnDeviceChange broadcast code 1HDDH1200100451 update type 0
[WARNING] Lidar sn: [1HDDH1200100451] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2020-12-20 16:07:28.086] [console] [info]  Recieve Message: Set 0 Id 7 Seq 75  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:28.086] [console] [info]  Send Command: Set 0 Id 2 Seq 6  [command_channel.cpp] [Send] [243]
[2020-12-20 16:07:28.088] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 5  [command_handler.cpp] [OnCommand] [118]
OnDeviceChange broadcast code 1HDDH3200100471 update type 0
[WARNING] Lidar sn: [1HDDH3200100471] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2020-12-20 16:07:28.088] [console] [info]  Send Command: Set 0 Id 2 Seq 9  [command_channel.cpp] [Send] [243]
[2020-12-20 16:07:28.110] [console] [info]  Update State to 4, device connect true  [device_manager.cpp] [UpdateDeviceState] [257]
OnDeviceChange broadcast code 1HDDH1200100451 update type 2
[WARNING] Lidar sn: [1HDDH1200100451] StateChange!!!
Device Working State 4
Device State Error Code 0X00000000
Device feature 0
[2020-12-20 16:07:28.110] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 6  [command_handler.cpp] [OnCommand] [118]
[... 27 lines cut ...]
[2020-12-20 16:07:29.406] [console] [info]  Recieve Message: Set 0 Id 7 Seq 88  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:29.502] [console] [info]  Recieve Message: Set 0 Id 7 Seq 89  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:29.598] [console] [info]  Recieve Message: Set 0 Id 7 Seq 90  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:29.694] [console] [info]  Recieve Message: Set 0 Id 7 Seq 91  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:29.790] [console] [info]  Recieve Message: Set 0 Id 7 Seq 92  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:29.886] [console] [info]  Recieve Message: Set 0 Id 7 Seq 93  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.006] [console] [info]  Recieve Message: Set 0 Id 7 Seq 94  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.102] [console] [info]  Recieve Message: Set 0 Id 7 Seq 95  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.198] [console] [info]  Recieve Message: Set 0 Id 7 Seq 96  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.294] [console] [info]  Recieve Message: Set 0 Id 7 Seq 97  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.390] [console] [info]  Recieve Message: Set 0 Id 7 Seq 98  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.486] [console] [info]  Recieve Message: Set 0 Id 7 Seq 99  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.606] [console] [info]  Recieve Message: Set 0 Id 7 Seq 100  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.703] [console] [info]  Recieve Message: Set 0 Id 7 Seq 101  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.798] [console] [info]  Recieve Message: Set 0 Id 7 Seq 102  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.895] [console] [info]  Recieve Message: Set 0 Id 7 Seq 103  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:30.991] [console] [info]  Recieve Message: Set 0 Id 7 Seq 104  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:31.086] [console] [info]  Recieve Message: Set 0 Id 7 Seq 105  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:31.206] [console] [info]  Recieve Message: Set 0 Id 7 Seq 106  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:31.303] [console] [info]  Recieve Message: Set 0 Id 7 Seq 107  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:31.399] [console] [info]  Recieve Message: Set 0 Id 7 Seq 108  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:31.495] [console] [info]  Recieve Message: Set 0 Id 7 Seq 109  [command_handler.cpp] [OnCommand] [121]
[2020-12-20 16:07:31.591] [console] [info]  Recieve Message: Set 0 Id 7 Seq 110  [command_handler.cpp] [OnCommand] [121]

There is a ping from Livox, but the data cannot be received.

Livox-SDK, 2020-12-21:
Hi @iSIlya 
According to Console log.

> [2020-12-20 16:07:31.591] [console] [info] Recieve Message: Set 0 Id 7 Seq 110 [command_handler.cpp] [OnCommand] [121]

LiDAR push some abnormal Information. Could you try to run `./lidar_sample` to see what exactly error code is.

iSIlya, 2020-12-21:
> LiDAR push some abnormal Information. Could you try to run `./lidar_sample` to see what exactly error code is.

Console log:
./lidar_sample
Livox SDK initializing.
Livox SDK has been initialized.
Livox SDK version 2.1.1 .
Start discovering device.
[2020-12-21 08:07:50.225] [console] [info]  Broadcast broadcast code: 1HDDH3200100471  [device_discovery.cpp] [OnBroadcast] [162]
Receive Broadcast Code 1HDDH3200100471
[2020-12-21 08:07:50.226] [console] [info] LocalIP: 192.168.1.209  [device_discovery.cpp] [OnBroadcast] [225]
[2020-12-21 08:07:50.226] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [226]
[2020-12-21 08:07:50.226] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [227]
[2020-12-21 08:07:50.252] [console] [info] New Device  [device_discovery.cpp] [OnData] [109]
[2020-12-21 08:07:50.252] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [110]
[2020-12-21 08:07:50.252] [console] [info] Broadcast Code: 1HDDH3200100471  [device_discovery.cpp] [OnData] [111]
[2020-12-21 08:07:50.252] [console] [info] Type: 3  [device_discovery.cpp] [OnData] [112]
[2020-12-21 08:07:50.252] [console] [info] IP: 192.168.1.212  [device_discovery.cpp] [OnData] [113]
[2020-12-21 08:07:50.252] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [114]
[2020-12-21 08:07:50.252] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [115]
[2020-12-21 08:07:50.253] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [241]
[2020-12-21 08:07:50.276] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [129]
[WARNING] Lidar sn: [1HDDH3200100471] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2020-12-21 08:07:50.276] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [241]
[2020-12-21 08:07:50.299] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [129]
firm ver: 6.10.0.0
[2020-12-21 08:07:50.369] [console] [info]  Update State to 1, device connect true  [device_manager.cpp] [UpdateDeviceState] [265]
[WARNING] Lidar sn: [1HDDH3200100471] StateChange!!!
Device Working State 1
Device State Error Code 0X00000800
Device feature 0
[2020-12-21 08:07:50.369] [console] [info]  Send Command: Set 0 Id 4 Seq 6  [command_channel.cpp] [Send] [241]
[2020-12-21 08:07:50.395] [console] [info]  Recieve Ack: Set 0 Id 4 Seq 6  [command_handler.cpp] [OnCommand] [129]
OnSampleCallback statue 0 handle 0 response 0 
data_type 2 packet num 100
data_type 2 packet num 200
data_type 2 packet num 300
data_type 2 packet num 400
data_type 2 packet num 500
data_type 6 packet num 600
[2020-12-21 08:07:50.605] [console] [info]  Broadcast broadcast code: 1HDDH1200100451  [device_discovery.cpp] [OnBroadcast] [162]
Receive Broadcast Code 1HDDH1200100451
[2020-12-21 08:07:50.605] [console] [info] LocalIP: 192.168.1.209  [device_discovery.cpp] [OnBroadcast] [225]
[2020-12-21 08:07:50.605] [console] [info] Command Port: 55502  [device_discovery.cpp] [OnBroadcast] [226]
[2020-12-21 08:07:50.605] [console] [info] Data Port: 56002  [device_discovery.cpp] [OnBroadcast] [227]
data_type 2 packet num 700
[2020-12-21 08:07:50.629] [console] [info] New Device  [device_discovery.cpp] [OnData] [109]
[2020-12-21 08:07:50.629] [console] [info] Handle: 1  [device_discovery.cpp] [OnData] [110]
[2020-12-21 08:07:50.629] [console] [info] Broadcast Code: 1HDDH1200100451  [device_discovery.cpp] [OnData] [111]
[2020-12-21 08:07:50.629] [console] [info] Type: 3  [device_discovery.cpp] [OnData] [112]
[2020-12-21 08:07:50.629] [console] [info] IP: 192.168.1.211  [device_discovery.cpp] [OnData] [113]
[2020-12-21 08:07:50.629] [console] [info] Command Port: 55502  [device_discovery.cpp] [OnData] [114]
[2020-12-21 08:07:50.629] [console] [info] Data Port: 56002  [device_discovery.cpp] [OnData] [115]
[2020-12-21 08:07:50.629] [console] [info]  Send Command: Set 0 Id 2 Seq 8  [command_channel.cpp] [Send] [241]
[2020-12-21 08:07:50.653] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 8  [command_handler.cpp] [OnCommand] [129]
[WARNING] Lidar sn: [1HDDH1200100451] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2020-12-21 08:07:50.653] [console] [info]  Send Command: Set 0 Id 2 Seq 9  [command_channel.cpp] [Send] [241]
data_type 2 packet num 800
[2020-12-21 08:07:50.677] [console] [info]  Update State to 4, device connect true  [device_manager.cpp] [UpdateDeviceState] [265]
[WARNING] Lidar sn: [1HDDH1200100451] StateChange!!!
Device Working State 4
Device State Error Code 0X00000000
Device feature 0
[2020-12-21 08:07:50.677] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 9  [command_handler.cpp] [OnCommand] [129]
[... 262 lines cut ...]
data_type 2 packet num 19000
data_type 6 packet num 19100
data_type 2 packet num 19200
[2020-12-21 08:07:57.469] [console] [info]  Recieve Message: Set 0 Id 7 Seq 91  [command_handler.cpp] [OnCommand] [132]
data_type 2 packet num 19300
data_type 2 packet num 19400
data_type 2 packet num 19500
[2020-12-21 08:07:57.589] [console] [info]  Recieve Message: Set 0 Id 7 Seq 92  [command_handler.cpp] [OnCommand] [132]
data_type 6 packet num 19600
data_type 2 packet num 19700
data_type 2 packet num 19800
[2020-12-21 08:07:57.685] [console] [info]  Recieve Message: Set 0 Id 7 Seq 93  [command_handler.cpp] [OnCommand] [132]
data_type 2 packet num 19900
data_type 2 packet num 20000
[... cut ...] ndler.cpp] [OnCommand] [132]
data_type 6 packet num 20100
data_type 2 packet num 20200
data_type 2 packet num 20300
[2020-12-21 08:07:57.877] [console] [info]  Recieve Message: Set 0 Id 7 Seq 95  [command_handler.cpp] [OnCommand] [132]
data_type 2 packet num 20400
data_type 2 packet num 20500
[2020-12-21 08:07:57.973] [console] [info]  Recieve Message: Set 0 Id 7 Seq 96  [command_handler.cpp] [OnCommand] [132]
data_type 6 packet num 20600
data_type 2 packet num 20700
data_type 2 packet num 20800