Livox-SDK/Livox-SDK #112: lidar_sample_cc  GetLidarDataCb 似乎调用不到？

opentld, 2021-03-24:
win10，VS2019 平台，Livox viewer可以正常启动MID-40，并显示点云数据，但运行lidar_sample_cc 如图：
[image]
应该是打开了雷达，但GetLidarDataCb 似乎调用不到？我应该在哪里获取点云数据进行处理呢？
按理说，应该在这里打印出来的，但是并没有打印数据
[image]
我在OnDeviceBroadcast里面加了一句LidarStartSampling调用，似乎也不好使？
[image]

我究竟该怎么做？才能取到点云数据自己进行处理？？？

opentld, 2021-03-24:
难道是p_lidar->connect_state = kConnectStateOff;的问题？不会吧？

opentld, 2021-03-29:
@Livox-SDK

Livox-SDK, 2021-03-31:
您好，@opentld 
首先，从控制台输出结果来看，没有打印收到 LiDAR 的广播包的日志，
结合 Livox Viewer 能看到点云的情况来看，
比较有可能是原因是 防火墙把 lidar_sample_cc 网络数据程序拦截了，
建议您在 windows defender 防火墙->允许的应用 中找到 lidar_sample_cc 并删除，
然后重新运行 lidar_sample_cc 程序

opentld, 2021-05-25:
> 您好，@opentld
> 首先，从控制台输出结果来看，没有打印收到 LiDAR 的广播包的日志，
> 结合 Livox Viewer 能看到点云的情况来看，
> 比较有可能是原因是 防火墙把 lidar_sample_cc 网络数据程序拦截了，
> 建议您在 windows defender 防火墙->允许的应用 中找到 lidar_sample_cc 并删除，
> 然后重新运行 lidar_sample_cc 程序

确实如此，防火墙的问题，谢谢！

lonlonago, 2021-09-05:
@opentld   您好，我也是同样的问题，麻烦加下我的微信交流下好吗？  394467238 ， 非常感谢~