Livox-SDK/Livox-SDK #114: 如何使用GPS对雷达进行时间同步？

YuQ-Zhou, 2021-04-06:
Windows系统 vs15 已经按照官方给的教程中将硬件设备连接好，并且使用livox viwever软件控制雷达时可以看到pps signal。

请问接下来的步骤是直接运行sdk中的lidar_utc_sync项目然后直接采集数据吗？

如果是的话该如何查看时间戳信息？我直接打印时间戳信息似乎是乱码？

feiwei93, 2021-04-22:
兄弟，你怎么打印的时间戳信息？乱码解决了吗？我在用GPS同步。

YuQ-Zhou, 2021-04-25:
打印出来了，可以看看官方给的livox扫描仪通信协议，里面有时间戳的存储格式。我的方法是前四位直接转int输出，后四位低位在前高位在后，先转成二进制补零，然后转回十进制

feiwei93, 2021-04-25:
解决了，感谢发自我的iPhone------------------ 原始邮件 ------------------发件人: YuQ-Zhou  ***@***.***>发送时间:

[Livox-SDK/Livox-SDK] 如何使用GPS对雷达进行时间同步？ (#114)

"target": "https://github.com/Livox-SDK/Livox-SDK/issues/114#issuecomment-826315772",
"url": "https://github.com/Livox-SDK/Livox-SDK/issues/114#issuecomment-826315772",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]