Livox-SDK/Livox-SDK #123: 如何使用SDK进行UTC时间同步

feiwei93, 2021-04-22:
如何utc同步来进行采样保存lvx文件呢？目前参考lidar_utc_sync与lidar_lvx_sample结合，控制逻辑是连接Lidar并配置同步，使用LidarStartSampling来进行采样，但收到的数据包时间戳并不是UTC时间。该如何在采样程序中嵌入同步UTC？能提供一下代码吗?

feiwei93, 2021-04-22:
@Livox-SDK

Gitwlmai, 2021-06-08:
请问你解决了吗？我也遇到同样问题，打印点云的时间戳与UTC时间戳不一致

Gitwlmai, 2021-06-10:
兄弟，我看你把这个问题关闭了，你是解决了吗？能分享一下吗？问题一直卡在这里

feiwei93, 2021-06-11:
不知道你用的是那种同步方式，我用的是GPS同步，你的是最新版的Livox-SDK吗？

来自 Outlook<http://aka.ms/weboutlook>

Gitwlmai, 2021-06-11:
我用的是GPS+PPS的同步方式，GPS发送UTC到我们的主机然后通过SDK发给雷达，PPS脉冲1HZ接雷达的同步接口，然后时间戳就如下所示
Rmc: $GNRMC,115611.00,A,2309.6336714,N,11322.8560422,E,0.013,319.4,090621,0.0,E,D*2A                                                                                     第一步接收RMC时间
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356419517 
    time: 21/6/9 11:3371000000                                                                                                                                                                                             打印RMC时间21/6/9 11:3371000000 
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356419917                                                                                                                                       点云输出时间 21-6-9 11:3356419917    （两者相差了将近15s）
    [2021-06-09 19:56:11.070] [console] [info]  Send Command: Set 1 Id 10 Seq 208  [command_channel.cpp] [Send] [247]
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356420316 
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356420715 
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356421114 
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356421515 
    lidar: data_type:2 ts_type:3 timeStamp: 21-6-9 11:3356421913 
    lidar: data_type:6 ts_type:3 timeStamp: 21-6-9 11:3356421623 
    [2021-06-09 19:56:11.072] [console] [info]  Recieve Ack: Set 1 Id 10 Seq 208  [command_handler.cpp] [OnCommand] [118]
    OnLidarSetRmcSyncTimeCallback statue 0 handle 0 response 0. 

   没有同步到点云上，有偏差，我用的是2.3.0的SDK，应该是最新的

feiwei93, 2021-06-11:
我的是先确认收到GPS信号，用GPS信号同步TX2本机时钟localtime，之后再运行lidar_utc_sync与lidar_lvx_sample的整合代码。
来自 Outlook<http://aka.ms/weboutlook>

Gitwlmai, 2021-06-11:
不是直接解析GPS端发过来的时间直接通过SDK给到雷达吗？

feiwei93, 2021-06-11:
你打印的rmc时间不是GNRMC的时间。你确定用GNRMC同步主机时间了吗？两者相差的时间可能是你代码运行的时间发自我的iPhone------------------ 原始邮件 ------------------发件人: Gitwlmai  ***@***.***>发送时间:

"target": "https://github.com/Livox-SDK/Livox-SDK/issues/123#issuecomment-859195069",
"url": "https://github.com/Livox-SDK/Livox-SDK/issues/123#issuecomment-859195069",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

feiwei93, 2021-06-11:
不是的，要先同步一下主机时间，你这个点云的时间不是GPS的时间，是主机的localtime，你看一下sample里面的lvx_file代码，确认一下是否用到过这个函数发自我的iPhone------------------ 原始邮件 ------------------发件人: Gitwlmai  ***@***.***>发送时间:

Gitwlmai, 2021-06-11:
我看了一下那个lidar_utc_sample，它是直接读取/dev/ttyUSB0发过来的GPRMC数据，然后解析出里面的UTC时间，直接通过SDK发给雷达，所以我是参照它这个sample做的，您说同步主机时间，官方提供的demo里面没看到这一步骤呢？

feiwei93, 2021-06-11:
那个lidar_utc_sample代码本身就不是完整的代码。你只运行了lidar_utc_sample这一个代码吗？这样吧，你先把主机的时间随便改成一个错误的时间，然后在运行你之前的代码，打印出来看看。
来自 Outlook<http://aka.ms/weboutlook>

Gitwlmai, 2021-06-11:
我大概理解您的意思，就是点云输出的时间戳应该是主机的localtime，我只需要根据GPS的时间修正主机的localtime就行了。但是我有个疑问，SDK为啥提供发送UTC时间的接口给lidar，然后官网的文档这样描述的：_**Livox 设备接收到 PPS 信号上升沿，并由 GPRMC 数据解析出正确的时间信息后，会设置点云时间为 GPS 时间，并保持此时间基准持续累加，来实现和 GPS 设备的时间同步**_  看它的意思是会设置点云时间为GPS时间。

Gitwlmai, 2021-06-11:
> 那个lidar_utc_sample代码本身就不是完整的代码。你只运行了lidar_utc_sample这一个代码吗？这样吧，你先把主机的时间随便改成一个错误的时间，然后在运行你之前的代码，打印出来看看。 来自 Outlook<http://aka.ms/weboutlook>
> […](#)

您好，我按照您的方法，将主机的时间修改之后，实际上点云的时间戳还是按照GPS时间进行输出的，只不过这个时间戳没有对齐，相差几秒。不知您那边是这么做的

tianyu-56, 2021-06-16:
> 我的是先确认收到GPS信号，用GPS信号同步TX2本机时钟localtime，之后再运行lidar_utc_sync与lidar_lvx_sample的整合代码。 来自 Outlook<http://aka.ms/weboutlook>
> […](#)
> ________________________________ 发件人: Fei Wei ***@***.***> 发送时间: [...]

您好，我最近也碰到了这个问题，我想先通过lidar_utc_sync进行雷达和UTC时间同步，然后使用lidar_lvx_sample保存lvx文件，但我发现两个不能同时运行，端口冲突，请问下您是通过什么方法解决的这个问题，把lidar_utc_sync和lidar_lvx_sample整合成一个代码运行吗？ 可以提供下参考吗  感谢！

feiwei93, 2021-06-18:
不好意思现在才看到消息，这是我整合后的完整代码，已经测试过可以运行希望对你有所帮助。把sample的原文件夹替换后重新编译。我在lvx_file.c中修改过生成lvx的path，你需要改成自己的路径。大家相互学习，欢迎交流！

来自 Outlook<http://aka.ms/weboutlook>

feiwei93, 2021-06-18:
> > 我的是先确认收到GPS信号，用GPS信号同步TX2本机时钟localtime，之后再运行lidar_utc_sync与lidar_lvx_sample的整合代码。 来自 Outlookhttp://aka.ms/weboutlook
> > […](#)
> > ________________________________ 发件人: Fei Wei _**@**_._**> 发送时间: 2021年6月11日 9:12 收件人: Livox-SDK/Livox-SDK _**@**_.**_> 主题: 回复: [Livox-SDK/Livox-SDK] 如何使用SDK进行UTC时间同步 (#123) 不知道你用的是那种同步方式，我用的是GPS同步，你的是最新版的Livox-SDK吗？ 来自 Outlookhttp://aka.ms/weboutlook
> 
> 您好，我最近也碰到了这个问题，我想先通过lidar_utc_sync进行雷达和UTC时间同步，然后使用lidar_lvx_sample保存lvx文件，但我发现两个不能同时运行，端口冲突，请问下您是通过什么方法解决的这个问题，把lidar_utc_sync和lidar_lvx_sample整合成一个代码运行吗？ 可以提供下参考吗 感谢！

你好，这是我整合后的代码，测试过可以运行，希望对你有所帮助，大家相互学习，欢迎交流！

#include <algorithm>
#include <string.h>
#include "lvx_file.h"
#include "cmdline.h"
#include <stdio.h>
#include <stdlib.h>
#include <thread>
#include <chrono>
#include "lds_lidar.h"
#include "synchro.h"

DeviceItem devices[kMaxLidarCount];
LvxFileHandle lvx_file_handler;
std::list<LvxBasePackDetail> point_packet_list;
std::vector<std::string> broadcast_code_rev;
std::condition_variable lidar_arrive_condition;
std::condition_variable extrinsic_condition;
std::condition_variable point_pack_condition;
std::mutex mtx;
int lvx_file_save_time =15; 
bool is_finish_extrinsic_parameter = false;
bool is_read_extrinsic_from_xml = false;
uint8_t connected_lidar_count = 0;

#define FRAME_RATE 20

using namespace std::chrono;

/** Cmdline input broadcast code */
static std::vector<std::string> cmdline_broadcast_code;

/** Connect all the broadcast device in default and connect specific device when use program options or broadcast_code_list is not empty. */
std::vector<std::string> broadcast_code_list = {
  "3GGDJ3300100351" 
};

/** Receiving error message from Livox Lidar. */
void OnLidarErrorStatusCallback(livox_status status, uint8_t handle, ErrorMessage *message) {
  static uint32_t error_message_count = 0;
  if (message != NULL) {
    ++error_message_count;
    if (0 == (error_message_count % 100)) {
      printf("handle: %u\n", handle);
      printf("temp_status : %u\n", message->lidar_error_code.temp_status);
      printf("volt_status : %u\n", message->lidar_error_code.volt_status);
      printf("motor_status : %u\n", message->lidar_error_code.motor_status);
      printf("dirty_warn : %u\n", message->lidar_error_code.dirty_warn);
      printf("firmware_err : %u\n", message->lidar_error_code.firmware_err);
      printf("pps_status : %u\n", message->lidar_error_code.device_status);
      printf("fan_status : %u\n", message->lidar_error_code.fan_status);
      printf("self_heating : %u\n", message->lidar_error_code.self_heating);
      printf("ptp_status : %u\n", message->lidar_error_code.ptp_status);
      printf("time_sync_status : %u\n", message->lidar_error_code.time_sync_status);
      printf("system_status : %u\n", message->lidar_error_code.system_status);
    }
  }
}

/** Receiving point cloud data from Livox LiDAR. */
void GetLidarData(uint8_t handle, LivoxEthPacket *data, uint32_t data_num, void *client_data) {
  if (data) {
    if (handle < connected_lidar_count && is_finish_extrinsic_parameter) {
[... 334 lines cut ...]
      std::unique_lock<std::mutex> lock(mtx);
      point_pack_condition.wait_for(lock, milliseconds(kDefaultFrameDurationTime) - (steady_clock::now() - last_time));
      last_time = steady_clock::now();
      point_packet_list_temp.swap(point_packet_list);
    }
    if(point_packet_list_temp.empty()) {
      printf("Point cloud packet is empty.\n");
      break;
    }
    printf("Finish save %d frame to lvx file.\n", i);
    lvx_file_handler.SaveFrameToLvxFile(point_packet_list_temp);
  }

  lvx_file_handler.CloseLvxFile();

  for (i = 0; i < kMaxLidarCount; ++i) {
    if (devices[i].device_state == kDeviceStateSampling) {
/** Stop the sampling of Livox LiDAR. */
      LidarStopSampling(devices[i].handle, OnStopSampleCallback, nullptr);
    }
  }

/** Uninitialize Livox-SDK. */
  Uninit();
}

feiwei93, 2021-06-18:
> > 我的是先确认收到GPS信号，用GPS信号同步TX2本机时钟localtime，之后再运行lidar_utc_sync与lidar_lvx_sample的整合代码。 来自 Outlookhttp://aka.ms/weboutlook
> > […](#)
> > ________________________________ 发件人: Fei Wei _**@**_._**> 发送时间: 2021年6月11日 9:12 收件人: Livox-SDK/Livox-SDK _**@**_.**_> 主题: 回复: [Livox-SDK/Livox-SDK] 如何使用SDK进行UTC时间同步 (#123) 不知道你用的是那种同步方式，我用的是GPS同步，你的是最新版的Livox-SDK吗？ 来自 Outlookhttp://aka.ms/weboutlook
> 
> 您好，我最近也碰到了这个问题，我想先通过lidar_utc_sync进行雷达和UTC时间同步，然后使用lidar_lvx_sample保存lvx文件，但我发现两个不能同时运行，端口冲突，请问下您是通过什么方法解决的这个问题，把lidar_utc_sync和lidar_lvx_sample整合成一个代码运行吗？ 可以提供下参考吗 感谢！

你好，这是我整合后的完整代码，已经测试过可以运行希望对你有所帮助。把sample的原文件夹替换后重新编译。我在lvx_file.c中修改过生成lvx的path，你需要改成自己的路径。大家相互学习，欢迎交流！

feiwei93, 2021-06-18:
> > 我的是先确认收到GPS信号，用GPS信号同步TX2本机时钟localtime，之后再运行lidar_utc_sync与lidar_lvx_sample的整合代码。 来自 Outlookhttp://aka.ms/weboutlook
> > […](#)
> > ________________________________ 发件人: Fei Wei _**@**_._**> 发送时间: 2021年6月11日 9:12 收件人: Livox-SDK/Livox-SDK _**@**_.**_> 主题: 回复: [Livox-SDK/Livox-SDK] 如何使用SDK进行UTC时间同步 (#123) 不知道你用的是那种同步方式，我用的是GPS同步，你的是最新版的Livox-SDK吗？ 来自 Outlookhttp://aka.ms/weboutlook
> 
> 您好，我最近也碰到了这个问题，我想先通过lidar_utc_sync进行雷达和UTC时间同步，然后使用lidar_lvx_sample保存lvx文件，但我发现两个不能同时运行，端口冲突，请问下您是通过什么方法解决的这个问题，把lidar_utc_sync和lidar_lvx_sample整合成一个代码运行吗？ 可以提供下参考吗 感谢！

你好，这是我整合后的完整代码，已经测试过可以运行希望对你有所帮助。大家相互学习，欢迎交流！
#include <algorithm>
#include <string.h>
#include "lvx_file.h"
#include "cmdline.h"
#include <stdio.h>
#include <stdlib.h>
#include <thread>
#include <chrono>
#include "lds_lidar.h"
#include "synchro.h"

DeviceItem devices[kMaxLidarCount];
LvxFileHandle lvx_file_handler;
std::list<LvxBasePackDetail> point_packet_list;
std::vector<std::string> broadcast_code_rev;
std::condition_variable lidar_arrive_condition;
std::condition_variable extrinsic_condition;
std::condition_variable point_pack_condition;
std::mutex mtx;
int lvx_file_save_time =10; //设置运行时间(s)
bool is_finish_extrinsic_parameter = false;
bool is_read_extrinsic_from_xml = false;
uint8_t connected_lidar_count = 0;

#define FRAME_RATE 20

using namespace std::chrono;

/** Cmdline input broadcast code */
static std::vector<std::string> cmdline_broadcast_code;

/** Connect all the broadcast device in default and connect specific device when use program options or broadcast_code_list is not empty. */
std::vector<std::string> broadcast_code_list = {
  "3GGDJ3300100351"
};

/** Receiving error message from Livox Lidar. */
void OnLidarErrorStatusCallback(livox_status status, uint8_t handle, ErrorMessage *message) {
  static uint32_t error_message_count = 0;
  if (message != NULL) {
    ++error_message_count;
    if (0 == (error_message_count % 100)) {
      printf("handle: %u\n", handle);
      printf("temp_status : %u\n", message->lidar_error_code.temp_status);
      printf("volt_status : %u\n", message->lidar_error_code.volt_status);
      printf("motor_status : %u\n", message->lidar_error_code.motor_status);
      printf("dirty_warn : %u\n", message->lidar_error_code.dirty_warn);
      printf("firmware_err : %u\n", message->lidar_error_code.firmware_err);
      printf("pps_status : %u\n", message->lidar_error_code.device_status);
      printf("fan_status : %u\n", message->lidar_error_code.fan_status);
      printf("self_heating : %u\n", message->lidar_error_code.self_heating);
      printf("ptp_status : %u\n", message->lidar_error_code.ptp_status);
      printf("time_sync_status : %u\n", message->lidar_error_code.time_sync_status);
      printf("system_status : %u\n", message->lidar_error_code.system_status);
    }
  }
}

/** Receiving point cloud data from Livox LiDAR. */
void GetLidarData(uint8_t handle, LivoxEthPacket *data, uint32_t data_num, void *client_data) {
  if (data) {
    if (handle < connected_lidar_count && is_finish_extrinsic_parameter) {
      std::unique_lock<std::mutex> lock(mtx);
[... 335 lines cut ...]
      std::unique_lock<std::mutex> lock(mtx);
      point_pack_condition.wait_for(lock, milliseconds(kDefaultFrameDurationTime) - (steady_clock::now() - last_time));
      last_time = steady_clock::now();
      point_packet_list_temp.swap(point_packet_list);
    }
    if(point_packet_list_temp.empty()) {
      printf("Point cloud packet is empty.\n");
      break;
    }
    printf("Finish save %d frame to lvx file.\n", i);
    lvx_file_handler.SaveFrameToLvxFile(point_packet_list_temp);
  }

  lvx_file_handler.CloseLvxFile();

  for (i = 0; i < kMaxLidarCount; ++i) {
    if (devices[i].device_state == kDeviceStateSampling) {
/** Stop the sampling of Livox LiDAR. */
      LidarStopSampling(devices[i].handle, OnStopSampleCallback, nullptr);
    }
  }

/** Uninitialize Livox-SDK. */
  Uninit();
}

tianyu-56, 2021-06-18:
非常感谢您！我好好学习一下，再次感谢！！

   

 ________________________________ 发件人: Fei Wei @.&gt; 发送时间: 2021年6月11日 9:12 收件人: Livox-SDK/Livox-SDK _@**._&gt; 主题: 回复: [Livox-SDK/Livox-SDK] 如何使用SDK进行UTC时间同步 (#123) 不知道你用的是那种同步方式，我用的是GPS同步，你的是最新版的Livox-SDK吗？ 来自 Outlookhttp://aka.ms/weboutlook
  
  
 #include <string.h&gt;
 #include <stdio.h&gt;
 #include <stdlib.h&gt;
 #include 
 
 std::list point_packet_list;
 
 
 
 
 
 printf("temp_status : %u\n", message-&gt;lidar_error_code.temp_status);
 printf("volt_status : %u\n", message-&gt;lidar_error_code.volt_status);
 printf("motor_status : %u\n", message-&gt;lidar_error_code.motor_status);
 printf("dirty_warn : %u\n", message-&gt;lidar_error_code.dirty_warn);
 printf("firmware_err : %u\n", message-&gt;lidar_error_code.firmware_err);
 printf("pps_status : %u\n", message-&gt;lidar_error_code.device_status);
 printf("fan_status : %u\n", message-&gt;lidar_error_code.fan_status);
 printf("self_heating : %u\n", message-&gt;lidar_error_code.self_heating);
 printf("ptp_status : %u\n", message-&gt;lidar_error_code.ptp_status);
 printf("time_sync_status : %u\n", message-&gt;lidar_error_code.time_sync_status);
 printf("system_status : %u\n", message-&gt;lidar_error_code.system_status);
 }
 }
 }
 
 if (handle < connected_lidar_count &amp;&amp; is_finish_extrinsic_parameter) {
 
 
 
 printf("OnGetLidarExtrinsicParameter statue %d handle %d response %d \n", status, handle, response-&gt;ret_code);
 lidar_info.pitch = response-&gt;pitch;
 lidar_info.roll = response-&gt;roll;
 lidar_info.yaw = response-&gt;yaw;
 lidar_info.x = static_cast(response-&gt;x / 1000.0);
 lidar_info.y = static_cast(response-&gt;y / 1000.0);
 lidar_info.z = static_cast(response-&gt;z / 1000.0);
 
 
 ack-&gt;firmware_version[0],
 ack-&gt;firmware_version[1],
 ack-&gt;firmware_version[2],
 ack-&gt;firmware_version[3]);
 }
 }
 
 uint8_t handle = info-&gt;handle;
 
 uint8_t handle = info-&gt;handle;
 
 uint8_t handle = info-&gt;handle;
 
 printf("OnDeviceChange broadcast code %s update type %d\n", info-&gt;broadcast_code, type);
 uint8_t handle = info-&gt;handle;
 if (handle &gt;= kMaxLidarCount) {
 
 printf("[WARNING] Lidar sn: [%s] Connect!!!\n", info-&gt;broadcast_code);
 printf("[WARNING] Lidar sn: [%s] Disconnect!!!\n", info-&gt;broadcast_code);
 printf("[WARNING] Lidar sn: [%s] StateChange!!!\n", info-&gt;broadcast_code);
 }
 
 
  
 if (info == nullptr || info-&gt;dev_type == kDeviceTypeHub) {
 return;
 } 
 
printf("Receive Broadcast Code %s\n", info-&gt;broadcast_code);
 (std::find(broadcast_code_rev.begin(), broadcast_code_rev.end(), info-&gt;broadcast_code) == broadcast_code_rev.end())) {
 broadcast_code_rev.push_back(info-&gt;broadcast_code);
 
 if ((steady_clock::now() - last_time + milliseconds(50)) &gt;= wait_time) {
 
 
 
 if ((broadcast_code_list.size() != 0) &amp;&amp;
 livox_status result = AddLidarToConnect(broadcast_code_rev[i].c_str(), &amp;handle);
 
  
 cmd.add("time", 't', "Time to save point cloud to the lvx file", false);
 cmd.parse_check(argc, const_cast<char **&gt;(argv));
 while ((pos = sn_list.find("&amp;")) != std::string::npos) {
 printf("Time to save point cloud to the lvx file:%d.\n", cmd.get("time"));
 lvx_file_save_time = cmd.get("time");
 }
 
 
LdsLidar&amp; read_lidar = LdsLidar::GetInstance();
 
 
 
Synchro&amp; synchro = Synchro::GetInstance();
 
 
 
 
 
 GetLivoxSdkVersion(&amp;_sdkversion);
 
 
 
  
 
 
 
 
 
 
 
 std::list point_packet_list_temp;
 {

tianyu-56, 2021-06-19:
非常抱歉又打扰您，我最近两天尝试编译了您发给我的代码，但我发现里面有很多错误，您发给我的应该是替换后的main.cpp，我已经把ample_cc\lidar_utc_sync下的lds_lidar和synchro代码复制到了sample\lidar_lvx_file文件夹下，运行还是会出错，请教下是我的过程有误吗，如果方便能在邮件附件里附上所有的文件吗？  十分抱歉浪费您的时间了！