Livox-SDK/Livox-SDK #124: PTP sync not working

filippocupolo, 2021-04-26:
Hi, following the instruction of [time synchronization](https://github.com/Livox-SDK/Livox-SDK/wiki/livox-device-time-synchronization-manual) I manged to install ptp4l and run the ptp sync with livox Avia. Looking at wireshark we can see that they are exchanging msgs.

[image]

The strange thing is that delay response msg is not sent.
Anyway looking at Livox viewer the AVIA has no sync

[image]

Test made with Ubuntu 18.04.5 and Livox Avia directly connected with ethernet (no router).

I made other test with Livox MID70 and Jetson Xavier (Jetpack 4.4) and with wireshark i can see all messages (Sync, Follow Delay_Req and also Delay_Resp) but still the lidar is not sync.

Any ideas on how could I solve this problem?

mbcel, 2021-05-23:
Any news on this? I am experiencing the same issue. However I do not even see the delay_req message. I also have a switch inbetween and dhcp mode enabled. I might give it a try with static ip the next days.

mbcel, 2021-05-24:
Okay I got it to work. However not with linuxptp as described in the wiki.

Instead I used the [ptpd](https://github.com/Livox-SDK/ptpd) project which is apparently is the project that is used by Livox.

Cloning and compiling the project and than running it with
`./src/ptpd -i eth0 --foreground -M`

worked for me.

However I am still puzzled why the linuxptp is not working. Opening the udp port 319, 320 also did not work for me.

secret104278, 2022-10-13:
Hi all, after digging into linuxptp, found this solution

1. Create a config file `my.cfg` as
```cfg
[global]
ignore_transport_specific 1
```

2. Start ptp4l with `-f my.cfg`
```
sudo ptp4l -i eno1 -m -f my.cfg
```

jukindle, 2023-02-10:
Hi @secret104278 , I also observed that with ptpd, synchronization works. However, even with your proposed config disabling transport specific checking, I am unable to get the LiDAR to synchronize to my host. Which version of ptp4l are you using, and, did you do any other specific configurations not mentioned here?