Livox-SDK/Livox-SDK #128: How to launch two devices

russellyq, 2021-06-02:
Hi, Thanks for your great work

I am trying to launch two devices at same time. Instead of using livox hub, are there any ways to launch two devices which are directly connected to laptop ?

I am using "roslaunch livox_ros_driver livox_lidar.launch multi_topic:="1" bd_list:="XX&XX""

However, it can only launch one of these two devices.

kamiccolo, 2021-06-03:
Indeed it seems like a restriction imposed by Livox-SDK itself. Just can have two different processes connecting to different LiDAR's in parallel. Having different port_count offset for different processes did not seem to work:
https://github.com/kamiccolo/Livox-SDK/commit/87a227b5ca9d71b94d3063e0dba1f83dcf5dc99b

I wonder if it has something to do with the bound and shared port for broadcasts?

TL;DR: having similar issue, looking for quick solution.

russellyq, 2021-06-03:
So there is no way but to use a livox hub to connect multiple devices ?

kamiccolo, 2021-06-03:
> So there is no way but to use a livox hub to connect multiple devices ?

well, currently there are at least 3 options:
 - reimplement the SDK functionallity 
 - fix the darn SDK
 - start those LiDAR's with a slight offset of time (this seems to work, at least most of the time). Like starting lidar-1, connecting to it, starting capture. Then starting lidar-2, connecting to it, starting capture, etc.
 
 Also, I wonder if something could be done using containers and/or separate networks for both LiDARs.

igricart, 2021-07-05:
I created each Livox in a different namespace. I could create two Livox Lidars with different frame_ids.

I made this change to the `livox_lidar.launch`
```
<launch>

	<arg name="lvx_file_path" default="livox_test.lvx"/>
	<arg name="bd_list" default="100000000000000"/>
	<arg name="xfer_format" default="0"/>
	<arg name="multi_topic" default="1"/>
	<arg name="data_src" default="0"/>
	<arg name="publish_freq" default="10.0"/>
	<arg name="output_type" default="0"/>
	<arg name="rviz_enable" default="false"/>
	<arg name="rosbag_enable" default="false"/>
	<arg name="cmdline_arg" default="$(arg bd_list)"/>
	<arg name="msg_frame_id" default="livox_frame"/>
	<arg name="node_name"/>
	<arg name="config_path"/>

	<param name="xfer_format" value="$(arg xfer_format)"/>
	<param name="multi_topic" value="$(arg multi_topic)"/>
	<param name="data_src" value="$(arg data_src)"/>
	<param name="publish_freq" type="double" value="$(arg publish_freq)"/>
	<param name="output_data_type" value="$(arg output_type)"/>
	<param name="cmdline_str" type="string" value="$(arg bd_list)"/>
	<param name="cmdline_file_path" type="string" value="$(arg lvx_file_path)"/>
	<param name="user_config_path" type="string" value="$(arg config_path)"/>
	<param name="frame_id" type="string" value="$(arg msg_frame_id)"/>
	
	<node name="$(arg node_name)" pkg="livox_ros_driver"
	      type="livox_ros_driver_node" required="true"
	      output="screen" args="$(arg cmdline_arg)"/>

	<group if="$(arg rviz_enable)">
		<node name="rviz" pkg="rviz" type="rviz" respawn="true"
				args="-d $(find livox_ros_driver)/config/display_lidar_points.rviz"/>
    </group>

	<group if="$(arg rosbag_enable)">
    	<node pkg="rosbag" type="record" name="record" output="screen"
          		args="-a"/>
    </group>

</launch>
```
And when calling in my launch file I use
```
  <group>
    <include file="$(find livox_ros_driver)/launch/livox_lidar.launch" ns="your_namespace_1">
      <arg name="node_name" value="your_node_name_1" />
      <arg name="multi_topic" value="0" />
      <arg name="msg_frame_id" value="livox_mid70_1" />
      <arg name="config_path" value="$(find amazing_package)/config/livox_1_config.json" />
    </include>
  </group>

  <group>
    <include file="$(find livox_ros_driver)/launch/livox_lidar.launch" ns="your_namespace_2">
      <arg name="node_name" value="your_node_name_2" />
      <arg name="multi_topic" value="0" />
      <arg name="msg_frame_id" value="livox_mid70_2" />
      <arg name="config_path" value="$(find amazing_package)/config/livox_2_config.json" />
    </include>
  </group>

```