Livox-SDK/Livox-SDK #134: I wanted to get the radar data of 0.2s

yuyu19970716, 2021-07-29:
Hello, when I was running lidar_lvx_sample, I found that I could only get the radar data in units of s, I wanted to get the radar data of 0.2s, then I looked in main. cpp and lvx_file. cpp to see where to modify, but I didn't find it, could you please tell me where to modify? Thank you very much!

lancewoo, 2021-08-03:
I'd suggest using gettimeofday() to do your own timing yourself in the data callback function. I did that.

yuyu19970716, 2021-08-03:
Hello!

 Thank you very much for your proposal! I also want to ask a question. I don't want to get radar data from the command line. I want to use the SDK library function in my own program to turn on the radar and get radar data in my own program, but when I do so, the SDK library function is displayed in my program, and many functions are not defined, I saw that only the function name in the library function did not complete the function implementation. At the same time, I found that my modification of the contents of the main. cpp or main.c function did not work. I don't know why. Do you have relevant experience? Thank you.

Best wishes to you!

yuyu19970716, 2021-08-03:
I want to use the library function of livox SDK and then use the function inside, but I don't know how to call it

lancewoo, 2021-08-04:
> 
> 
> I want to use the library function of livox SDK and then use the function inside, but I don't know how to call it

It seems you have to run up the sample application first things first.