Livox-SDK/Livox-SDK #135: How can I tranlate an intensity value to an RGB color ?

lancewoo, 2021-07-31:
@Livox-SDK  Hi, I just wonder how can I tranlate an intensity value to an RGB color ?
Say, I have a LivoxExtendRawPoint.
https://github.com/Livox-SDK/Livox-SDK/blob/05cc408e54bce90863cd4f160992a92ddea1fd29/sdk_core/include/livox_def.h#L186

How do I calculate the color value as Livox Viewer shows？
Thanks!

[image]

Livox-SDK, 2021-08-13:
Hi lancewoo,

I am sorry to let you know that; Lidar is not able to detect the color of object; Only camara does;

lancewoo, 2021-08-14:
> 
> 
> Hi lancewoo,
> 
> I am sorry to let you know that; Lidar is not able to detect the color of object;
> To detect the color the camera need to be involved ;

You didn't answer my question. My question is how do you convert a reflectivity intensity value to an RGB value in your Livox Viewer? I didn't mean the colors of objects that Lidar scans.