Livox-SDK/Livox-SDK #147: Not able to capture Lidar data from Livox Mid 70 connected to Rasberry Pi 4, using Python code

nikhilpTF, 2021-09-24:
Hi,

We are trying to capture lidar data from Livox Mid 70 sensor from Rasberry Pi 4, with Python library openpylivox (https://github.com/Livox-SDK/openpylivox) 
We have connected the LAN cable from Livox to the Raspberry Pi ethernet port and ran below code (we assigned a static IP to the sensor).

sensor.auto_connect("172.25.25.100") 

Note: 172.25.25.100 is the IP address of Rasberry Pi

We are getting this error:
   self._cmdSocket.sendto(connect_request, (self._sensorIP, 65000)) OSError: [Errno 22] Invalid argument 

Would be very grateful if anyone can help to get this working..

gsoga, 2022-02-04:
Hi @nikhilpTF ,

I don't know if you managed to get it and I don't know that library, but in case someone finds it useful you can check the user manual (https://www.livoxtech.com/mid-70/downloads) and to try the default config:
 "The default subnet masks of the Mid-70 sensors are all 255.255.255.0, and their default gateways are 192.168.1.1."

For example, your IP address could be 192.168.1.5 with the submask 255.255.255.0

I hope it's useful

DonFreiling, 2022-11-03:
I am having the same issue with a Livox Tele-15. The address of the Tele-15 is 192.168.2.196. I am running this on a Jetson Xavier. I've set the IP address for eth0 to 192.168.2.1 with netmask 255.255.255.0. When run the demo I get:
airm@CalFixture:~/devl/OpenPyLivox$ python livox_controller_demo.py 
Using computer IP address: 127.0.1.1

   *** Discovered a Livox sensor ***
           Type: Tele-15
         Serial: 1PQDH760010296
          IP(s): 192.168.2.196 (ID: 1)
                 
Using computer IP address: 127.0.1.1

   *** Discovered a Livox sensor ***
           Type: Tele-15
         Serial: 1PQDH760010296
          IP(s): 192.168.2.196 (ID: 1)
                 
Attempting to auto-connect to the Livox Tele-15 with S/N: 1PQDH760010296
Traceback (most recent call last):
  File "livox_controller_demo.py", line 263, in <module>
    singleSensorDemo()
  File "livox_controller_demo.py", line 46, in singleSensorDemo
    connected = sensor.auto_connect()
  File "/home/airm/devl/OpenPyLivox/openpylivox/openpylivox.py", line 2220, in auto_connect
    numFound = self.connect(self._computerIP, lidarSensorIPs[0], 0, 0, 0)
  File "/home/airm/devl/OpenPyLivox/openpylivox/openpylivox.py", line 2080, in connect
    self._cmdSocket.sendto(connect_request, ("192.168.2.196", 65000))
OSError: [Errno 22] Invalid argument

Any suggestions?