Livox-SDK/Livox-SDK #154: Lidar Imu transformation matrix

aditdoshi333, 2021-11-16:
Hello,

Is there any way to access the calibration matrix between IMU and lidar from livox sdk or driver? And is there any way to know that the calibration is disturbed? 

Thank you

Livox-SDK, 2022-06-12:
Hi aditdoshi333

Sorry, the  calibration matrix is not able to be accessed.

kakusikun, 2024-04-29:
how to get the extrinsic parameters like this?

https://github.com/Livox-SDK/LIO-Livox/blob/master/launch/horizon.launch#L15

Marselka, 2024-06-03:
@Livox-SDK could you comment please where lidar to built-in IMU extrinsics are usually obtained from?

I use mid 360 and information UDP packets do not contain such extrinsics. No information in User Manual too. Livox Viewer 2 also lacks extrinsics.

Looks that at least raw extrinsics should be in access somewhere.

5uvin, 2025-04-09:
@Livox-SDK Is it still the case that lidar to IMU extrinsics are not publicly available?

RicoJia, 2026-01-19:
> how to get the extrinsic parameters like this?
> 
> https://github.com/Livox-SDK/LIO-Livox/blob/master/launch/horizon.launch#L15

I believe an adjacent file https://github.com/Livox-SDK/LIO-Livox/blob/master/launch/mid360.launch should give a good estimate