Livox-SDK/Livox-SDK #156: Processing changes

soccken, 2021-11-29:
Currently I am doing measurements with LiDAR using the lidar_lvx_file generated in / build / sample in the Livox-SDK file.
When this file is executed by loop processing, the processing time of the program is slow because the inisialize SDK and ADD listening device are executed every time the loop is executed.
Is it possible to loop only the process of getting point cloud data as an lvx file while the initialization and device addition are continued?

gabrielheider, 2022-05-05:
Hi @soccken ,
Did you solve this issue? I have same problem! 
Also I do not understand how to change working mode. If you know how, could you provide a sample example?
Thanks

soccken, 2022-05-16:
Hi,gabrielheider.
I haven't solved the problem either. 
 Is there any solution?

gabrielheider, 2022-05-18:
Hi soccken,
I have solved it partly. I wanted to save the points to a .txt file, that works now as fast as it work for the lvx file. If you are interested in that code tell me and i will send/post it.

Changing the working mode works like that:
// change lidar mode to power saving
    livox_status status = LidarSetMode(devices[0].handle, kLidarModePowerSaving, OnChangeWorkingModeCallback, nullptr);
    printf("Livox Status return %i \n", status);
    Sleep(200);

OnChangeWorkingModeCallback is basicaly an empty funktion:
/** Callback function of change working mode. */
void OnChangeWorkingModeCallback(livox_status status, uint8_t handle, uint8_t response, void* data) {
    printf("Working state: %d handle %d response %d \n", status, handle, response);
    if (status == kStatusSuccess) {
        printf("changed working state sucessfully \n");
        SetDeviceStateUpdateCallback(OnDeviceInfoChange);
    }
    else if (status == kStatusTimeout) {
        printf("timeout");
    }
}
It seems to me that it doesn't work wenn you do more than one state or mode change in a short time.  So to avoid this you can use Sleep(200);

soccken, 2022-05-18:
Which file has this modification?  
Also, does it mean that only the part that gets the lvx file is looped by modifying it?