Livox-SDK/Livox-SDK #159: I can't receive data from Livox mid 70 with ubuntu 20.04

Kiseikinoyama, 2021-12-28:
Register broadcast code: 3GGDJ9600100401
Time to save point cloud to the lvx file:10.
Livox SDK initializing.
Livox SDK has been initialized.
Livox SDK version 2.3.0 .
Start discovering device.
[2021-12-28 13:49:13.974] [console] [info]  Broadcast broadcast code: 3GGDJ9600100401  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 3GGDJ9600100401
[2021-12-28 13:49:13.974] [console] [info] Broadcast code : 3GGDJ9600100401 not add to connect  [device_discovery.cpp] [OnBroadcast] [163]
[2021-12-28 13:49:16.995] [console] [info]  Broadcast broadcast code: 3GGDJ9600100401  [device_discovery.cpp] [OnBroadcast] [150]
Receive Broadcast Code 3GGDJ9600100401
[2021-12-28 13:49:16.996] [console] [info] LocalIP: 192.168.1.50  [device_discovery.cpp] [OnBroadcast] [201]
[2021-12-28 13:49:16.996] [console] [info] DeviceIP: 192.168.1.140  [device_discovery.cpp] [OnBroadcast] [202]
[2021-12-28 13:49:16.996] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [203]
[2021-12-28 13:49:16.996] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [204]
[2021-12-28 13:49:16.998] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
[2021-12-28 13:49:16.999] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [100]
[2021-12-28 13:49:16.999] [console] [info] Broadcast Code: 3GGDJ9600100401  [device_discovery.cpp] [OnData] [101]
[2021-12-28 13:49:16.999] [console] [info] Type: 6  [device_discovery.cpp] [OnData] [102]
[2021-12-28 13:49:16.999] [console] [info] IP: 192.168.1.140  [device_discovery.cpp] [OnData] [103]
[2021-12-28 13:49:16.999] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [104]
[2021-12-28 13:49:16.999] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [105]
[2021-12-28 13:49:16.999] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [243]
[2021-12-28 13:49:17.000] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [118]
OnDeviceChange broadcast code 3GGDJ9600100401 update type 0
[WARNING] Lidar sn: [3GGDJ9600100401] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2021-12-28 13:49:17.001] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [243]
[2021-12-28 13:49:17.002] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [118]
firm ver: 10.7.0.0
[2021-12-28 13:49:17.054] [console] [info]  Update State to 0, device connect true  [device_manager.cpp] [UpdateDeviceState] [281]
OnDeviceChange broadcast code 3GGDJ9600100401 update type 2
[WARNING] Lidar sn: [3GGDJ9600100401] StateChange!!!
Device Working State 0
Device State Change Progress 0
Device feature 0
[2021-12-28 13:49:20.881] [console] [info]  Device 0 removed   [device_manager.cpp] [RemoveDevice] [97]
OnDeviceChange broadcast code 3GGDJ9600100401 update type 1
[WARNING] Lidar sn: [3GGDJ9600100401] Disconnect!!!

closed 2022-01-06 by Kiseikinoyama

byliutao, 2022-04-30:
如果你用的是hub_lvx_sample，可以试一下lidar_lvx_sample

dariojavo, 2022-08-13:
Hi @Kiseikinoyama, may I ask how you manage to solved this problem?

Yadunund, 2022-09-30:
@dariojavo I had a similar issue which I documented here https://github.com/Livox-SDK/Livox-SDK/issues/186
I resolved it by using a better power supply.