Livox-SDK/Livox-SDK #164: Livox SDK for Windows

WingmanMedia, 2022-02-21:
I thought Livox SDK is for Linux and ROS platforms only but then I have noticed on the following page there is one for Windows. However when I click on learn more it sends me to the SDK related to Linux. Where I can see the one related to Windows?

https://www.livoxtech.com/sdk 

Do not get me wrong but I have 0 experience using Linux and I am not really keen to learn it. 
However I have 5 years experience writing web  applications  and Windows apps on C#. Even those skills are rusty(I left IT 12 years ago) but at least using it under windows  will be in the environment I know and it will be much easier for me to remember and learn.

Livox-SDK, 2022-06-12:
Hi WingmanMedia

Compile SDK does not have to under Linux environment. You can refer to the description in the SDK-Readme file to compile SDK:
[image]

tomas1515, 2024-06-27:
Is Visual Studio 17 2022 compilation working? Will a DDL library be created at compile time?

F2M360, 2024-09-16:
What's the latest version of visual studio for 2022, 2023 or 2024. The last one was for 2019 and I'm have some issues with this at the moment